jketterl/openwebrx #175: Problem With SDRPlay API V 3.07.02 and OpenWebRX v 20.0

Challado, 2020-08-31:
**Bug Description**
I upgrade my OpenWEBRX v. 19, SoapySDR v7 to v 20-dev and SoapySDR v8 using the API v 3.06.01. All works fine at first time.
I decide to upgrade the API too, to v 3.07.02. When I do these upgrade, OpenWebRX stops to open the API server, indicating the following error.

Aug 31 10:23:01 sdr openwebrx[12525]: #033[1m#033[31m[ERROR] error in activateStream() - Init() failed: sdrplay_api_Fail#033[0m
Aug 31 10:23:01 sdr openwebrx[12525]: client connection establised

When I start the SoapyRemote and use CubicSDR to connect to device, all works fine with the same API version, on the same device, that makes me believe that the problem can be on OpenWebRX start string or something like that.

**To Reproduce**
Upgrade Openwebrx V 19, SoapySDR v7 and API to Openwebrx V 20-Dev, SoapySDR v8 and API v 3.07.02

**Expected behavior**
Start and works OpenWebRX correctly.

**Installation method**
Manual Installation

**Versions**
OpenWebRX version v0.20.0-dev, latest commit

**Log messages**
Aug 31 09:23:52 sdr openwebrx[11882]: 2020-08-31 09:23:52,599 - owrx.__main__ - INFO - OpenWebRX version v0.20.0-dev starting up...
Aug 31 09:23:52 sdr openwebrx[11882]: 2020-08-31 09:23:52,659 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: SDRPLAY RSP1A
Aug 31 09:25:03 sdr openwebrx[11882]: 2020-08-31 09:25:03,637 - owrx.connection - DEBUG - client connection intitialized
Aug 31 09:25:03 sdr openwebrx[11882]: 2020-08-31 09:25:03,647 - owrx.source - DEBUG - activating profile 0
Aug 31 09:25:04 sdr openwebrx[11882]: 2020-08-31 09:25:04,629 - owrx.source - INFO - Started sdr source: soapy_connector -s 500000 -a "Antenna A" -P 0 -g IFGR=20,RFGR=8 -f 240000 -p 46661 -c 34973 -d driver=sdrplay -t if_mode=0
Aug 31 09:25:04 sdr openwebrx[11882]: setting up control socket...
Aug 31 09:25:04 sdr openwebrx[11882]: control socket started on 46661
Aug 31 09:25:04 sdr openwebrx[11882]: socket setup complete, waiting for connections
Aug 31 09:25:04 sdr openwebrx[11882]: [INFO] devIdx: 0
Aug 31 09:25:04 sdr openwebrx[11882]: [INFO] hwVer: 255
Aug 31 09:25:04 sdr openwebrx[11882]: Setting gain element IFGR: 20.000000 dB
Aug 31 09:25:04 sdr openwebrx[11882]: Setting gain element RFGR: 8.000000 dB
Aug 31 09:25:04 sdr openwebrx[11882]: [INFO] Using format CS16.
Aug 31 09:25:04 sdr openwebrx[11882]: #033[1m#033[31m[ERROR] error in activateStream() - Init() failed: sdrplay_api_Fail#033[0m
Aug 31 09:25:04 sdr openwebrx[11882]: client connection establised
Aug 31 09:25:04 sdr openwebrx[11882]: 2020-08-31 09:25:04,732 - owrx.source.connector - DEBUG - opening control socket...
Aug 31 09:25:04 sdr openwebrx[11882]: control connection established
Aug 31 09:25:04 sdr openwebrx[11882]: 2020-08-31 09:25:04,733 - owrx.dsp - DEBUG - received STATE_RUNNING, attempting DspSource restart
Aug 31 09:25:04 sdr openwebrx[11882]: 2020-08-31 09:25:04,884 - owrx.fft - DEBUG - Spectrum thread initialized successfully.
Aug 31 09:25:04 sdr openwebrx[11882]: 2020-08-31 09:25:04,884 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 46661 | csdr fft_cc 8192 3267.97385620915 | csdr logaveragepower_cf -70 8192 17 | csdr fft_exchange_sides_ff 8192 | csdr compress_fft_adpcm_f_u8 8192
Aug 31 09:25:04 sdr openwebrx[11882]: client connection establised
Aug 31 09:25:04 sdr openwebrx[11882]: Connection to 127.0.0.1 46661 port [tcp/*] succeeded!
Aug 31 09:25:05 sdr openwebrx[11882]: WARNING: SoapySDRDevice_readStream timeout!
Aug 31 09:25:05 sdr openwebrx[11882]: 2020-08-31 09:25:05,992 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 46661 | csdr shift_addfast_cc --fifo /tmp/openwebrx/openwebrx_pipe_shift_pipe_3042528304 | csdr fir_decimate_cc 41 0.0036585365853658534 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx/openwebrx_pipe_bpf_pipe_3042527856 0.02624 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx/openwebrx_pipe_squelch_pipe_3042528944 --outfifo /tmp/openwebrx/openwebrx_pipe_smeter_pipe_3042528496 5 2 | csdr amdemod_cf | csdr fastdcblock_ff | csdr fractional_decimator_ff 1.016260162601626 | csdr agc_ff --profile slow --initial 200 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
Aug 31 09:25:06 sdr openwebrx[11882]: 2020-08-31 09:25:06,002 - owrx.dsp - DEBUG - adding new output of type audio
Aug 31 09:25:06 sdr openwebrx[11882]: 2020-08-31 09:25:06,003 - owrx.dsp - DEBUG - adding new output of type smeter
Aug 31 09:25:06 sdr openwebrx[11882]: fir_decimate_cc: taps_length = 1093
Aug 31 09:25:06 sdr openwebrx[11882]: csdr bandpass_fir_fft_cc: csdr fir_decimate_cc: taps_length = 1093
Aug 31 09:25:06 sdr openwebrx[11882]: csdr fir_decimate_cc: padded_taps_length = 1096
Aug 31 09:25:06 sdr openwebrx[11882]: csdr fir_decimate_cc: taps = 121e168
Aug 31 09:25:06 sdr openwebrx[11882]: csdr fir_decimate_cc: NEON aligned taps = 121e180
Aug 31 09:25:06 sdr openwebrx[11882]: fifo control mode on
Aug 31 09:25:06 sdr openwebrx[11882]: Connection to 127.0.0.1 46661 port [tcp/*] succeeded!
Aug 31 09:25:06 sdr openwebrx[11882]: csdr shift_addfast_cc: fifo control mode on
Aug 31 09:25:06 sdr openwebrx[11882]: client connection establised
Aug 31 09:25:06 sdr openwebrx[11882]: csdr fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
Aug 31 09:25:06 sdr openwebrx[11882]: csdr fractional_decimator_ff: not using taps
Aug 31 09:25:06 sdr openwebrx[11882]: csdr squelch_and_smeter_cc: fifo control mode on
Aug 31 09:25:06 sdr openwebrx[11882]: AGC PARAMS:
Aug 31 09:25:06 sdr openwebrx[11882]:   hang_time = 600
Aug 31 09:25:06 sdr openwebrx[11882]:   reference = 0.800000
Aug 31 09:25:06 sdr openwebrx[11882]:   attack_rate = 0.010000
Aug 31 09:25:06 sdr openwebrx[11882]:   decay_rate = 0.000100
Aug 31 09:25:06 sdr openwebrx[11882]:   max_gain = 65536.000000
Aug 31 09:25:06 sdr openwebrx[11882]:   initial_gain = 200.000000
Aug 31 09:25:06 sdr openwebrx[11882]:   attack_wait_time = 0
Aug 31 09:25:06 sdr openwebrx[11882]:   gain_filter_alpha = 1.500000
Aug 31 09:25:06 sdr openwebrx[11882]: csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 153) + (input_size = 360) - 1
Aug 31 09:25:06 sdr openwebrx[11882]: (overlap_length = 152) = taps_length - 1
Aug 31 09:25:06 sdr openwebrx[11882]: csdr shift_addfast_cc: reinitialized to -0
Aug 31 09:25:06 sdr openwebrx[11882]: csdr bandpass_fir_fft_cc: filter initialized, low_cut = -0.328, high_cut = 0.328
Aug 31 09:25:06 sdr openwebrx[11882]: csdr squelch_and_smeter_cc: initial squelch level is 1e-15
Aug 31 09:25:06 sdr openwebrx[11882]: WARNING: SoapySDRDevice_readStream timeout!
Aug 31 09:25:07 sdr openwebrx[11882]: WARNING: SoapySDRDevice_readStream timeout!
Aug 31 09:25:08 sdr openwebrx[11882]: WARNING: SoapySDRDevice_readStream timeout!
Aug 31 09:25:09 sdr openwebrx[11882]: WARNING: SoapySDRDevice_readStream timeout!

**Additional context**
Add any other context about the problem here.

jketterl, 2020-08-31:
It's still an SDRPlay problem. The error messages do not originate from OpenWebRX or the soapy_connector, but from either the SoapySDRPlay module or the SDRPlay server itself. I really don't know why you insist bringing this in here, it's not my fault that other parties are providing different levels of support.

The things I can see:
- Your frequency is 240000, but your sample rate is 500000. That means that the frequency range will span into the negative range. I cannot tell if that is a problem or not, I haven't tried it myself.
- Your `RFGR=8` is higher than the limit of any SDRPlay device I know. If your device name is correct and this is really an RSP1, I think the maximum is `RFGR=3`. Please cross-check with SoapySDRUtil.

Challado, 2020-08-31:
If the problem is SoapySDR, agree with me that isn't work with SoapySDRemote too? 
The start frequency is only a example.  Wth higher frequencies same problem.
Follow the SoapySDRutil --probe. We have RFGR from 0 to 9.

Otherwise, if I downgrade the API version, all works ok.

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
[INFO] devIdx: 0
[INFO] hwVer: 255

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=19110D0B97
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=255

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 48] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 9] dB
  Full freq range: [0.001, 2000] MHz
    RF freq range: [0.001, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.125, 0.25, 0.5, 1, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

jketterl, 2020-08-31:
I cannot tell what's working and what isn't. That's because the error message is not very verbose and doesn't really tell me what's wrong. Since it's not originating from my code, it's hard for me to tell what's working and what isn't.

What device is this, really? I do remember changing the default configuration at some point because `RFGR=3` was the max for the RSP1.

You may as well check the output of the sdrplay service (`sudo journalctl -u sdrplay`) for more information. But that is clearly beyond scope here.

If it works in version 3.06, that's also an indicator that this is a problem with the API drivers. Have you updated your SoapySDRPlay module to go along? Which SoapySDRPlay module are you using? I think the recommended one to go with API version 3 is this one here: https://github.com/SDRplay/SoapySDRPlay

Challado, 2020-08-31:
> I cannot tell what's working and what isn't. That's because the error
> message is not very verbose and doesn't really tell me what's wrong. Since
> it's not originating from my code, it's hard for me to tell what's working
> and what isn't.
Really the API need to be more verbose to explain why not have started,
with a error or something like that.

> What device is this, really? I do remember changing the default
> configuration at some point because RFGR=3 was the max for the RSP1.
RSP1a.

> You may as well check the output of the sdrplay service (sudo journalctl
> -u sdrplay) for more information. But that is clearly beyond scope here.
Aug 31 10:20:54 sdr sdrplay_apiService[12408]: [12408]: sdrplay_apiService:
sdrplay_apiServiceWorkerThread: Entry
Aug 31 10:20:54 sdr sdrplay_apiService[12408]: [12409]:
sdrplay_apiService_device: heartBeatThread: Entry
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12422]:
sdrplay_apiService_device: mCmdHandlerThread: Entry
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12408]: sdrplay_apiService:
SelectDevice: ERROR: InitialiseMappingTableFromFile error -4: invalid hw
version 255
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12423]:
sdrplay_apiService_rsp: streamThread: Entry
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12422]:
sdrplay_apiService_rsp_log_enable 3
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12408]:
sdrplay_apiService_log_enable: 0
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12422]:
sdrplay_apiService_rsp_log_enable 0
Aug 31 10:21:24 sdr sdrplay_apiService[12408]: [12423]:
sdrplay_apiService_rsp: streamThread: Exit
Aug 31 10:21:24 sdr sdrplay_apiService[12408]: [12422]:
sdrplay_apiService_device: mCmdHandlerThread: Exit
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_device: mCmdHandlerThread: Entry
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12432]:
sdrplay_apiService_rsp: streamThread: Entry
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_rsp_log_enable 0
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12408]:
sdrplay_apiService_log_enable: 0
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_rsp_log_enable 0
Aug 31 10:21:32 sdr sdrplay_apiService[12408]: [12408]:
sdrplay_apiService_log_enable: 0
Aug 31 10:21:32 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_rsp_log_enable 0
Aug 31 10:22:20 sdr sdrplay_apiService[12408]: [12432]:
sdrplay_apiService_rsp: streamThread: Exit
Aug 31 10:22:20 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_device: mCmdHandlerThread: Exit
Aug 31 10:23:01 sdr sdrplay_apiService[12408]: [12685]:
sdrplay_apiService_device: mCmdHandlerThread: Entry
Aug 31 10:23:01 sdr sdrplay_apiService[12408]: [12686]:
sdrplay_apiService_rsp: streamThread: Entry
Aug 31 10:23:01 sdr sdrplay_apiService[12408]: [12685]:
sdrplay_apiService_rsp_log_enable 3
Aug 31 10:20:54 sdr sdrplay_apiService[12408]: [12408]: sdrplay_apiService:
sdrplay_apiServiceWorkerThread: Entry
Aug 31 10:20:54 sdr sdrplay_apiService[12408]: [12409]:
sdrplay_apiService_device: heartBeatThread: Entry
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12422]:
sdrplay_apiService_device: mCmdHandlerThread: Entry
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12423]:
sdrplay_apiService_rsp: streamThread: Entry
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12422]:
sdrplay_apiService_rsp_log_enable 3
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12408]:
sdrplay_apiService_log_enable: 0
Aug 31 10:21:23 sdr sdrplay_apiService[12408]: [12422]:
sdrplay_apiService_rsp_log_enable 0
Aug 31 10:21:24 sdr sdrplay_apiService[12408]: [12423]:
sdrplay_apiService_rsp: streamThread: Exit
Aug 31 10:21:24 sdr sdrplay_apiService[12408]: [12422]:
sdrplay_apiService_device: mCmdHandlerThread: Exit
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_device: mCmdHandlerThread: Entry
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12432]:
sdrplay_apiService_rsp: streamThread: Entry
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_rsp_log_enable 0
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12408]:
sdrplay_apiService_log_enable: 0
Aug 31 10:21:25 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_rsp_log_enable 0
Aug 31 10:21:32 sdr sdrplay_apiService[12408]: [12408]:
sdrplay_apiService_log_enable: 0
Aug 31 10:21:32 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_rsp_log_enable 0
Aug 31 10:22:20 sdr sdrplay_apiService[12408]: [12432]:
sdrplay_apiService_rsp: streamThread: Exit
Aug 31 10:22:20 sdr sdrplay_apiService[12408]: [12431]:
sdrplay_apiService_device: mCmdHandlerThread: Exit
Aug 31 10:23:01 sdr sdrplay_apiService[12408]: [12685]:
sdrplay_apiService_device: mCmdHandlerThread: Entry
Aug 31 10:23:01 sdr sdrplay_apiService[12408]: [12686]:
sdrplay_apiService_rsp: streamThread: Entry
Aug 31 10:23:01 sdr sdrplay_apiService[12408]: [12685]:
sdrplay_apiService_rsp_log_enable 3

> If it works in version 3.06, that's also an indicator that this is a
> problem with the API drivers. Have you updated your SoapySDRPlay module to
> go along? Which SoapySDRPlay module are you using? I [...]
Yep. Recompiled all from source and using the git code from these link that
you have posted.

I back to API v3.6 for now and all works.

jketterl, 2020-08-31:
I can pinpoint the location of the error message in SoapySDRPlay: https://github.com/SDRplay/SoapySDRPlay/blob/master/Streaming.cpp#L322

The output of the sdrplay service looks similar to mine, even the error about the hardware version (RSPduo here). Can't really see anything wrong there.

Even if the probe says RFGR range is 0-9, would you mind giving it a shot with a lower value from 0-3? It's my best guess right now...

jketterl, 2020-08-31:
Oh, two more things I just noticed from the probe:

- The "antenna" setting is not in agreement, it should be "RX". Usually this is not a problem, but maybe there's a check in the newest version.
- I'm not sure if you specified an "if_mode" in the config, but it's in the `soapy_connector` command, but doesn't seem to be supported according to the probe. Would you mind checking and removing it from the config if necessary?

Challado, 2020-08-31:
Test done and... BINGO.
Accept IFGR=20,RFGR=9 on v3.6, won't on v3.7.02. No other parameters are
modified. Appears really a API bug. I'll open a report on SDRPlay git site.

jketterl, 2020-08-31:
Yes, please. Seems like the probe is reporting incorrect values then. Maybe it doesn't recognize the RSP1A correctly.

loewal, 2020-08-31:
My values:root@Pi4:~ # SoapySDRUtil  --probe="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=B0001P0002
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=1, type=string, options=(0, 1, 2, 3)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 42] dB
    **IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 3] dB**
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

Challado, 2020-08-31:
Your API is 2.13. Mine is 3.07.02

jketterl, 2020-08-31:
I think he just meant to confirm the range when probed with the old driver. That's what I remembered, too.

loewal, 2020-08-31:
👍

jketterl, 2020-09-10:
I'm going to close this. I think this has produced sufficient results to solve the problem. Until the API will correctly respond with an error, I can't really offer a software fix for this, so for now this is something to consider during configuration.

closed 2020-09-10 by jketterl