jketterl/openwebrx #177: Integration of the data decoder rtl_433 to debug errors with sensors in the ism bnd

hubertmeier, 2020-09-17:
Hi,

there is a data decoder on github which can decode the sensors like temperature sensors for radio controlled clocks or other components in the ISM Bands. It would be easy to see if a sensor transmits and if we see the right sensor transmitting if the data would be decoded. most of the sensors show temperature and other wheater data, others like homematic-sensors are encrypted which can't be decoded.

It seems that the software calls the rtl_sdr directly, Openwebrx needs software which can accept audio, so it is perhaps impossible ... 

Thank you for your work on openwebrx

jketterl, 2020-09-17:
OpenWebRX can deliver pretty much any kind of data, but is not able to mimick a USB device, so it won't work. There's the possibility of faking an rtl_tcp socket, but that has other complications, the biggest one being that the application will assume it is able to control rtl-sdr parameters over it, whereas in reality, this won't be allowed.

Also, the integration of this would require some new frontend panels that are able to display the available information. I think it would be kind of hard to offer this in a webpage, but it may as well be feasible. I'm not familiar with the structure of the messages present.

I'll have to label this as an "idea" since the idea behind the "feature" label is that that type of issue is basically only waiting to be implemented, whereas issues labeled with "idea" need some critical input.

hubertmeier, 2020-09-17:
thank you, marking as idea for the future is fine, obviously is a file as input possible too.

openwebrx is a perfect companion to a kiwisdr and after your work much more capable than the original from ha7ilm. the idea behind openwebrx is absolutely cool ...

jketterl, 2022-07-13:
Looks like it's possible to feed IQ data into rtl_433: https://github.com/merbanan/rtl_433/issues/2110

TODO: figure out sample rate and best filter setup (how wide is that ISM band anyway?)

seems to be sufficient to run some experiments.

jketterl, 2023-09-12:
This is available in the latest development version now. The display of results may need some work, please leave feedback here.