jketterl/openwebrx #178: Display additional information of Digital Radio mondiale DRM

hubertmeier, 2020-09-22:
Hi,

DRM seems to work fine on the different test receivers with openwebrx 0.20.dev, on the kiwisdr there is a display with additional informations about different parameters of DRM, is it possible on openwebrx too?

Thank you for your work, its fun to play with it.

jketterl, 2020-09-22:
If you know how to get that information out of stock dream, I'd be happy to implement it. So far, all I'm getting out of it is audio.

The guys at KiwiSDR have patched dream to their needs, but I am not willing to go down that route, especially on a abandoned project like dream.

jketterl, 2020-11-04:
I will need to label this as an idea since it is not actionable right now. If anybody has any input on this, don't hesitate to leave it here.

xxj737, 2025-09-16:
I hope to add DRM functionality

Display additional information

jketterl, 2025-10-24:
Please make sure that modifications are upstreamed to the official project so that they can be picked up by distributions. It would probably also be a good idea to put together a new release. I don't intend to supply forked or modified versions because that will cause a lot of conflicts and support issues in the field.

Alternatively, you'll have to convince the linux community that you're taking over development of the dream project, and make sure the distributions switch to your version.