jketterl/openwebrx #180: Adjust frequency offset in software

jwt27, 2020-09-27:
After installing an upconverter, I find that I now have to tune slightly higher on SSB signals.  The offset is about 80-120 Hz, depending on profile.  You'd think this would be simple to fix, just add 80-120 Hz to `lfo_offset` for the affected profiles.  But then I discover that the RTL-SDR is not accurate enough to tune to this exact frequency.  I tried many different combinations of `lfo_offset` and `center_freq`, and while the offset does jump around a bit, there is no point where it is exactly right.

So, another feature request:  Would it be possible to add an option to adjust for such an error in software?  It would be similar to what `lfo_offset` is now, but only correct the frequency and scale displayed in the web UI.  Or, if having two very similar options is undesirable, maybe the meaning of `lfo_offset` can be changed so that it only applies a negative offset to the displayed frequency, so then `center_freq` is always the exact frequency that the SDR device is tuned to, and small errors can be corrected without relying on hardware accuracy.

jketterl, 2020-09-27:
There's already two settings for that purpose, `lfo_offset` and `ppm`. I think you can understand that I'm not too excited at the prospect of adding another.

jwt27, 2020-09-27:
I get that, but `lfo_offset` is currently unusable for such small offsets, at least on the RTL-SDR.  Hence why my request is to change the meaning of this setting to be only a software correction.

Also `ppm` only takes whole numbers so it's not useful for <1 ppm correction.  In my specific case it doesn't appear to be a frequency-relative error, anyway.

jketterl, 2020-09-27:
Well it is a software-adjustment. You'll have to counteract with your center_freq so that in combination with the correct lfo_offset the actual sdr frequency is hitting one of the steps.

To this day I wasn't even aware that there is a stepsize.

This setting has already been implemented the opposite way (i.e. you'd configure the SDR on the frequency it receives on its input, but the displayed frequency was offset), but it was not intuitive. A setting that influences the display also has to be implemented in way more places. The idea here is to correct the frequency once, and anything that comes beyond that does not need to know about the offset.

jketterl, 2020-09-27:
This is definitely possible with the existing parameters. When thinking about it: All that new parameter would do would be to manipulate the frequency parameters, among them the `center_freq`. So, manipulating the `center_freq` along with the correct counter-correction of the `lfo_offset` will give you the results you're looking for. And, that way of configuring it is even more reasonable because that way you retain the "true" frequencies in your configuration (and everywhere else).

jwt27, 2020-09-27:
Okay, I think I see what you mean.  I did the following for my 80-60m profile: (for reference I use the volmet at 3413 and the carrier of an AM station at 4800)

With HDSDR I find a center frequency that is at the boundary between two tuning steps. I find one at 129,264,000, then at this LO the error appears to be +85 Hz.  Then I add this number to `lfo_offset` and subtract it from `center_freq`, so my settings for OWRX become: `"center_freq": 4263915` and `"lfo_offset": 125000085`.

That does work, the offset is gone now.  But it's quite finicky, and I'll have to do this for every profile.  From a user-friendlyness perspective it would be a lot easier if `lfo_offset` only alters the displayed frequency and not where the hardware is tuned to.  But I understand that will make the code more complicated and difficult to follow.

Also using an odd number for `center_freq` doesn't play nice with my "snap to grid" modification, I'll have to look into that...

Somewhat related, there is this experimental branch for the RTL-SDR software that has some commits related to tuner accuracy:
https://github.com/keenerd/rtl-sdr
Unfortunately it's rather old and doesn't work with owrx_connector.

Anyhow, I have a solution now, so feel free to close this if you don't want to pursue this idea further.

jketterl, 2020-11-04:
I am aware of a whole bunch of different versions of the rtl-sdr library. Unfortunately, I can only rely on the core functionality that is available in the version that is provided by most distributions, which is usually the osmocom branch. I have heard complaints before that they are slow to merge features, but there's not much I can do about that...

As for the actual problem... I'd like to leave it at that. I do agree it's not the prettiest solution, but it is the one that is closer to what's actually happening from a technical perspective. As I said, the LFO compensation used to be on the rendering end before, and it creates a pretty big mess. It may be doable with a separate abstraction layer at a later point. I'll keep this in mind.

closed 2020-11-04 by jketterl