jketterl/openwebrx #181: Why dongles serialization doesn't work?

T-Shilov, 2020-09-28:
Hello dear **Jakob**!

I am using OpenWebRX **0.19.1** and **Debian10 / 64**.

I have several dongles. For some unknown reason, they sometimes spontaneously change places on the bands.
To fix this, I flashed their serial numbers:

**Serial number: 00000001
Serial number: 00000002
...................................................
Serial number: 00000006**

and included them in the config file:

```
.............................................................
sdrs = {

    "rtlsdr_1": {
        "name": "Ch1:",
        "type": "rtl_sdr",
        "ppm": 0,
        "antenna": "Collinear Dual Band",
        "profiles": {
            "10m": {
                "device": "serial=00000001",
                "name": "28 MHz",
                "center_freq": 28000000,
                "rf_gain": 8.7,
                "samp_rate": 2400000,
                "start_freq": 27175000,
                "start_mod": "nfm",
            },
        },
    },

    "rtlsdr_2": {
        "name": "Ch2:",
        "type": "rtl_sdr",
        "ppm": 0,
        "antenna": "Collinear Dual Band",
        "rf_gain": 1.4,
        "profiles": {
            "2m": {
                "device": "serial=00000002",
                "name": "145 MHz",
                "center_freq": 145000000,
                "samp_rate": 2400000,
                "start_freq": 145725000,
                "start_mod": "nfm",
            },
        },
    },

.................................................

    "rtlsdr_6": {
        "name": "Ch6:",
        "type": "rtl_sdr",
        "ppm": 0,
        "antenna": "Collinear Dual Band",
        "profiles": {
            "70cm": {
                "device": "serial=00000006",
                "name": "446 MHz",
                "center_freq": 446000000,
                "rf_gain": "33.8",
                "samp_rate": 2400000,
                "start_freq": 446000000,
                "start_mod": "nfm",
            },
        },
    },
}
.............................................................
```

However, this does not work because an error occurs:

"This receiver is currently unavailable due to technical issues.
Error Message: No SDR Devices available"

Why does this error occur?

jketterl, 2020-09-28:
If this is a question, this is the wrong place to ask. If this is a but report, use the issue template!

closed 2020-09-28 by jketterl

T-Shilov, 2020-09-28:
Yes, that sounds like a question. Sorry, but I don't know where to ask this, so I asked in Issue.
Jacob, tell me, please, how to ask a question correctly?

UB0ALH, 2020-09-29:
> Yes, that sounds like a question. Sorry, but I don't know where to ask this, so I asked in Issue.
> Jacob, tell me, please, how to ask a question correctly?

T-Shilov, you should write a letter to openwebrx@groups.io to ask question. I didn't know about it at the beginning too  )))

UB0ALH, 2020-09-29:
Read https://groups.io/g/openwebrx/topic/76985203 
I had the same question

UB0ALH, 2020-10-01:
;)