jketterl/openwebrx #184: Improvement UI: display a zoomed range (like 14000000 - 14350000) when changing the preset / band

hubertmeier, 2020-10-09:
A nice feature would be the possibility to give a zoom range which is displayed when the preset / band of the SDR is changed.

2m is easy, i choose a sample rate of 2 MSamples and choose the middle of the band, so i have automatically displayed 144-146MHz. If i want to choose a small band like 11m or amateur bands, the band range is relatively small and the samplerate is much higher, which is ok. It would be nice to have an initial zoom range to show only the small band, a zoom out to the sample rate is also possible ... or there is a additional parameter to fix the zoom range or to permit the zoom out.

In the config, there must be 3 additional parameters:

    zoom-lower-end,
    zoom-higher-end
    zoom-out-of-range: possible or not

jketterl, 2020-11-04:
Future development will probably cover this. The idea is to provide the ability to use I/Q sub-sampling to extract multiple "child" profiles that would then even be available in parallel. This will of course work for a single child profile as well and should cover this.

cptechnik, 2021-05-17:
I did it until now to reduce the samplerate, 
but at the new versions i can't go deeper as 1MS unless to get bullshit on waterfall and audio.

So all the small bands ...all shortwave Bands... are smaller than 2 MHz ...especially if somebody wants to use a RTL stick to receive 60,12 Meter or... god bless... freenet , 149 MHz...

For the first time its enough to set a start-zoom level, like Gmaps or Openstreetmap  ... z=0 to ...z=13...
or...
for the pre-first time... could this be done by javascript...  push 5 times zoom-plus...?