jketterl/openwebrx #187: Improvement UI: installing a button to clear the message window

hubertmeier, 2020-10-23:
I would be nice to have a button to clear the message window (output digtal decoders). After changing the mode FT4, FT8, WSPR and so on the new messages apear directly under the old ones. 
If a button is not possible there could be a blank line or a horzontal line if we change the decoder.

Thank you

jketterl, 2020-10-23:
I have told you to use the issue template before, yet still you completely ignore that. Result: by not listing a target audience, I will assume that you're the single person to benefit from this feature, which makes it pretty much the lowest priority. Happy now?

hubertmeier, 2020-10-25:
i dont know what i have done wrong, but its ok. that should be a feature request, it is not personal and i did not found this in another post. This was written in the android App Openhub, there is no template.  

thank you

jketterl, 2020-10-25:
Well, it's good to know that the workflow I have set up for this repository is easily subverted by some third-party app. Please don't use it to report issues on my repositories in the future.

jketterl, 2020-12-09:
This has been implemented in the current development / experimental version. If you have a testing setup, feel free to try it out. Otherwise, this will be in the next release.

closed 2021-01-14 by jketterl