jketterl/openwebrx #189: Version 0.20.0 does not display avatar on www.receiverbook.de

T-Shilov, 2020-11-02:
**Describe the bug**
Version 0.20.0 does not display its openwebrx-avatar.png avatar in the list on https://www.receiverbook.de/

**To Reproduce**
Go to the website https://www.receiverbook.de/ and see the list of versions.

**Expected behavior**
Avatars must be displayed in any version of OpenWebRX

**Installation method**
Debian Package 10/64.

**Versions**
0.20.0

**Log messages**
I do not know.

**Additional context**
In old versions (0.19.1, 0.18.0) and newer 0.21.0-dev, the avatar openwebrx-avatar.png is displayed successfully.

jketterl, 2020-11-02:
Sorry, but custom avatars are currently not a feature of OpenWebRX.

closed 2020-11-02 by jketterl

T-Shilov, 2020-11-02:
Sorry, but in older versions this feature worked fine. Why did you turn it disable? :-O

jketterl, 2020-11-02:
It has never been a feature, at least not as long as I have been working on OpenWebRX. Older versions had documentation to simply swap out the file, but I don't consider that a feature. This old "method" conflicts with the package manager and is therefore unsupported.

T-Shilov, 2020-11-03:
Well, you can't solve the avatar problem in OpenWebRX. 
Then maybe you can solve it by improving the site https://www.receiverbook.de ?
After all, you made this site, right? It really lacks avatars and other useful information.

On the other hand on the sites http://www.websdr.org and http://www.kiwisdr.com  there is a lot of useful information, and this is very convenient for their users.

jketterl, 2020-11-03:
I'm not solving any problems that don't exist. If you keep insisting, I will remove the avatars from receiverbook.

You have had my word now, you can either decide to live with it, or you can choose to keep insisting and bring disadvantages to others, it's your call. I will however not shift my priorities at your will.