jketterl/openwebrx #191: jquery.nanoscroller.js contains non existing sourcemap ref

alexander-pick, 2020-11-04:
The file  jquery.nanoscroller.js of the ubuntu install package contains a source map reference at the end to a not existing source map. 

```
//# sourceMappingURL=jquery.nanoscroller.js.map
```

This causes a warning in Chrome. I looked in the source version here and it has the same reference. After removing it the warning was gone.

jketterl, 2020-11-04:
This is a non-fatal error, I guess. It has been in the application for quite some time (longer than I am working on the application). I suggest to ignore it.

closed 2020-11-04 by jketterl

alexander-pick, 2020-11-04:
The original lib is distributed with the map file. You can simply remove this comment in your local copy. From what I saw the lib wasn't updated in 6yrs anyways so I don't expect any update problem here and the warning is gone. It still causes a 404 browser request.

jketterl, 2020-11-04:
Yes, I tried adding the map file (it wasn't that easy to get, https://jamesflorentino.github.io/nanoScrollerJS/ only gives you the .js files, but not the maps), but since I included the lib in a concatenated JS request, it didn't work that way. I swapped out the library with the minified version and removed the map reference.

The 404 request should only occur upon opening the developer tools (there's no need for the map during normal use), so this really wasn't that much of an issue.

This has been added a long time ago. I should probably look for a replacement at some point.

alexander-pick, 2020-11-04:
You can just remove that (//# sourceMappingURL=jquery.nanoscroller.js.map) line, should work better. The map file is not needed at all.

jketterl, 2020-11-04:
I have :wink: It doesn't show in the diff since I did that on the new file right away.

commit e700f0a9 (2020-11-04): replace nanoscroller with compressed version, drop map reference.
closes #191