jketterl/openwebrx #192: Unsupported soundcard mode causes application to hang

alexander-pick, 2020-11-04:
An soundcard with a not supported mode will cause an unhandled exception in the receiver.js (AudioEngine.js):

```
receiver.js:3946 Uncaught ReferenceError: targetRate is not defined
    at AudioEngine.setupResampling (receiver.js:3946)
    at new AudioEngine (receiver.js:3763)
    at openwebrx_init (receiver.js:1422)
    at onload ((index):32)
``` 

The var targetRate istused in the divlog error messages but it seems it is undefined. This will cause the whole application to hang and the waterfall won't load. You will be stuck with the background and the header. 

I encountered the issue by accidently setting my headset to handfree mode which seems to run in a unsupported mode and could only see the error by digging in the developer tools console for it.

jketterl, 2020-11-04:
Interesting. I have never had the opportunity to do much testing in this scenario since I am unable to reproduce it...

jketterl, 2020-11-04:
Alright, this was the result of some refactoring some while ago. The commit above should fix the JavaScript error, it can however not compensate for the unsupported sound card sample rate. You will see an error message in log window in the browser, but you still won't get any audio because the server is not able to provide the audio in the corresponding sample rate.

Just out of curiosity, what sample rate are you running when the error appears?

closed 2020-12-10 by jketterl