jketterl/openwebrx #196: Onboard WiFi disabled on fresh install of Raspberry Pi image

doismellburning, 2020-11-29:
**Describe the bug**
I imaged an SD card with the latest stable Raspberry Pi image (2020-10-11-OpenWebRX-full from https://www.openwebrx.de/download/rpi.php).
After imaging but before putting the card in my Pi, I dropped "standard Pi config files" in `/boot` - empty `ssh` and also `wpa_supplicant.conf`:

```
$ cat wpa_supplicant.conf
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="lorem"
    psk="ipsum"
}
```

I put the card in my Pi 3B with onboard WiFi, but couldn't see it appear on the network.

I then plugged in a WiFi dongle and was able to SSH in ok, but got a login message of:

```
Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.
```

and indeed:

```
$ rfkill
ID TYPE      DEVICE      SOFT      HARD
 0 wlan      phy0     blocked unblocked
 1 wlan      phy1   unblocked unblocked
 2 bluetooth hci0   unblocked unblocked
```

**To Reproduce**

* Image a fresh SD card with the RPi image
* Add `wpa_supplicant.conf` to `/boot` before booting
* Boot the Pi
* Discover it is not connecting to WiFi
* Connect via cable / a WiFi dongle
* Observe successful connection, and rfkill message on SSH

**Expected behavior**

I expect either WiFi to Just Work (though I'm not sure how viable that is) OR notes about it being initially disabled in the Installation docs

**Installation method**
How did you install OpenWebRX? (Raspberry Pi SD card image, Debian / Ubuntu packages, Docker image, manually?)

Raspberry Pi SD card image

**Versions**
What version of OpenWebRX are you running? (Check on startup, or see `owrx/version.py`. If a `-dev` version is used, ideally state the commit the issue is appearing on)

2020-10-11-OpenWebRX-full

**Log messages**
Are there any relevant messages relating to the bug in the output / log of OpenWebRX? (On most installations, the log should be available using the command `sudo journalctl -u openwebrx`)

```
Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.
```

**Additional context**
Add any other context about the problem here.

jketterl, 2020-11-29:
I'm sorry, but the way the WiFi behaves is inherited from standard Raspbian. I know there has been various issues and misunderstandings about "rfkill" (which mainly just ensures that you don't use the WiFi without having it set to the correct country before), but if it's not working this is out of my hands. The official Raspbian documentation applies to the OpenWebRX images, so I'd suggest those for research.

I will not implement custom WiFi handling.

doismellburning, 2020-11-29:
That seems reasonable and I wouldn't expect you to. What's the contribution process for the Wiki / installation docs? "Just go edit", or something else?

jketterl, 2020-11-29:
Normally, it's just edit, but to be honest, this doesn't belong into the OpenWebRX documentation. It doesn't have anything to do with OpenWebRX.

doismellburning, 2020-11-29:
> It doesn't have anything to do with OpenWebRX.

I'm not sure I'd agree.

I appreciate it's a property of the underlying OS, but from a user perspective, it's "I flashed an SD card and configured it for my WiFi with good configuration on a good Pi, but the OpenWebRX image doesn't work".

It seems about as applicable as the other "few notes about the images" in https://github.com/jketterl/openwebrx/wiki/Setup-Guide.

To that end, I've added it there - obviously it's your Wiki and you can remove it if you feel it doesn't belong, but I'd like to be able to recommend this to other people without having to add my own caveats to the install docs...!

Meanwhile, now I have it working, it looks excellent, thank you very much!

closed 2020-11-29 by doismellburning

loewal, 2020-11-29:
Please use raspi-config to set your country.

jketterl, 2020-11-29:
Well then, it's in there now, let's leave it as long as nobody complains. It's just a difficult situation because where would the line be? Installing a desktop environment? Setting up VNC? All those may be useful to some in conjunction with OpenWebRX, but they would be pretty lengthy, and there's sufficient documentation about them already.

Maybe the documentation should state more clearly that the base system is Raspbian and link to its documentation.

doismellburning, 2020-11-29:
> Well then, it's in there now, let's leave it 

Thank you

> where would the line be

I appreciate your point but I think there's a reasonable line around "can I plug it in and will it work" - you don't need to "[install] a desktop environment" or "[set] up VNC" to *use* OpenWebRX - they are, as you say, "useful to some in conjunction"

Meanwhile, until I'd figured out why I couldn't connect, this issue was a barrier to me _using_ the software at all - hence I'd be rather keen to see the line remain!

> Maybe the documentation should state more clearly that the base system is Raspbian and link to its documentation.

That probably wouldn't hurt, but I still think it's worth explicitly calling this issue out in the install docs, as above

Hopefully that all makes sense, cheers! :)