jketterl/openwebrx #199: SDRplay RSPduo and independent RX mode

righthalfplane, 2020-12-14:
The SDRplay RSPduo can independently tune the two channels, but you need to use the correct branch of SoapySDRPlay -

https://github.com/SDRplay/SoapySDRPlay/tree/dual-tuner-submodes

when you build this branch you get five device choses - the independent one is labeled -

"SDRplay Dev2 RSPduo **** - Dual Turner (independent RX)"

Here is a you.tube example of SdrGlut using the independent RX mode on the Raspberry Pi 4 -

https://youtu.be/17eJI8KIkYg

jketterl, 2020-12-14:
Please use the issue templates. I will start closing issues without any further reaction if you cannot even be bothered to invest a few minutes of time to make my life easier. If you had followed the template, you would have found out that "support for multi-channel sdrs (RSPduo in master/slave mode)" is already on the roadmap, so there's no need for this separate issue.

I do own RSPDuo devices and I am aware of their dual tuner capabilities. The current implementations makes it extremely hard to set up, but you can use both inputs at the same time with current OpenWebRX already, but since it is very error-prone I haven't included the setup process in the documentation. This is with the "master" branch of the repository you posted, I'm not sure what additional changes the branch contains.

Also, in our experimentation, it turned out to be somewhat pointless since a Raspberry Pi 4 is choking on CPU and cannot provide the two streams simultaneously. Analysis showed that the bottleneck lies in the SDRPlay API service, so not much we can change about that. This becomes even weirder when you get to see the same Raspberry Pi 4 handling two separate RSPDuos without any problems.

Nevertheless, there is efforts underway to make the underlying connector framework support multiple IQ channels, so that devices with multiple channels can be supported in the future. It will take some time to complete since this is a bigger change, so don't hold your breath.

closed 2020-12-14 by jketterl

righthalfplane, 2020-12-23:
The branch that I posted is the result of three weeks of work between me and the developer. It has the code that lets the two turners actually tune independently. Until that branch get pulled into the main branch - it needs to be used. Duel turner mode in the main branch aways controls both turners and they are not independent.