jketterl/openwebrx #207: log total client connected

IU1JFE, 2021-01-04:
_Originally posted by @jketterl in https://github.com/jketterl/openwebrx/issues/204#issuecomment-754117302_

hi, i have need to keep track of users connection variation for statistic and monitoring usage for best fine setting tuning .

for example ,usign the following command ,i can read easly the data using script and update usage plot :
sudo cat /var/log/syslog | grep "Total client"

example output:
Jan  4 16:08:41 raspberrypi openwebrx[892]: 2021-01-04 16:08:41,889 - owrx.client - INFO - Total client connected 0
Jan  4 17:26:07 raspberrypi openwebrx[892]: 2021-01-04 17:26:07,960 - owrx.client - INFO - Total client connected 1
Jan  4 17:32:53 raspberrypi openwebrx[892]: 2021-01-04 17:32:53,794 - owrx.client - INFO - Total client connected 2
Jan  4 17:33:43 raspberrypi openwebrx[892]: 2021-01-04 17:33:43,966 - owrx.client - INFO - Total client connected 1
Jan  4 17:33:43 raspberrypi openwebrx[892]: 2021-01-04 17:33:43,972 - owrx.client - INFO - Total client connected 1
Jan  4 17:37:29 raspberrypi openwebrx[892]: 2021-01-04 17:37:29,815 - owrx.client - INFO - Total client connected 0
Jan  4 17:37:29 raspberrypi openwebrx[892]: 2021-01-04 17:37:29,816 - owrx.client - INFO - Total client connected 0
Jan  4 17:53:21 raspberrypi openwebrx[892]: 2021-01-04 17:53:21,909 - owrx.client - INFO - Total client connected 1

jketterl, 2021-01-04:
Number of clients is exposed through the `/metrics` API / URL. Please use that instead of misusing the logs.

https://github.com/jketterl/openwebrx/wiki/API#httpyour-urlip8073metrics

closed 2021-01-04 by jketterl

IU1JFE, 2021-01-04:
> 
> 
> Number of clients is exposed through the `/metrics` API / URL. Please use that instead of misusing the logs.
> 
> https://github.com/jketterl/openwebrx/wiki/API#httpyour-urlip8073metrics

ok ,tanks, but using `/metrics ` API i need tro cstant monitoring the api, with ny mod i keep track on log file when user connect or disconnect, and using `sudo cat /var/log/syslog | grep "Total client" | sed 's/^.*Total client connected //'` i could get data evrey 12h.

is ther another way to do that pourpouse ?
tanks

jketterl, 2021-01-04:
No, there isn't, OpenWebRX is not a monitoring system, that's why this is exposed the way it is.

I suggest you get some basic information about data collection tools like collectd or telegraf, and other associated systems like rrdtool, grafana, cacti... There's plenty of choices out there.