jketterl/openwebrx #208: Multiple RTL-SDR USB Dongles

ch0mik, 2021-01-09:
Hi

Im adding the device 1 and 2 for sdrs/rtlsdr. Information from : https://github.com/jketterl/openwebrx/issues/93

'''
sdrs = {
    "rtlsdr": {
        "name": "RTL-SDR 2m",
        "type": "rtl_sdr",
        "ppm": 0,
		"device": 1,
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        "lfo_offset": -13000,
        "profiles": {
            "2m": {
                "name": "2m całe pasmo",
                "center_freq": 145000000,
                "rf_gain": 99,
                "samp_rate": 2048000,
                "start_freq": 145525000,
                "start_mod": "nfm",
            },
        },
            "rtlsdr": {
        "name": "RTL-SDR 70cm",
        "type": "rtl_sdr",
        "ppm": 0,
		"device": 2,
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        "lfo_offset": -13000,
        "profiles": {
            "70cm": {
                "name": "70cm Przemienniki",
                "center_freq": 438800000,
                "rf_gain": 99,
                "samp_rate": 2400000,
                "start_freq": 438350000,
                "start_mod": "nfm",
            },
        },
    }
}
'''

but after restart i have error ms on web : 

**
This receiver is currently unavailable due to technical issues.
Error Message:
No SDR Devices available
**

jketterl, 2021-01-09:
If you wanted to report a bug, please follow the issue template. This is not a bug however, so please contact the mailing list.

closed 2021-01-09 by jketterl

ch0mik, 2021-01-09:
my mistake .. it works now ;)