jketterl/openwebrx #209: OpenWebRX on OpenWrt

najdanovicivan, 2021-01-09:
**Feature description**
I'd like to be able to run the OpenWebRX on OpenWRT. I can start working on porting the required packages.

**Target audience**
Anyone willing to run this on OpenWrt 

Are there any python related dependencies since python is split in multiple packages on OpenWrt

D0han, 2021-01-09:
Not sure if i understood correctly, are you asking for this?
```
$ git grep -h 'import ' -- '*.py' | sort -u
from abc import ABC, ABCMeta, abstractmethod
from abc import ABC, abstractmethod
from abc import ABCMeta
from abc import ABCMeta, abstractmethod
from .admin import AdminController
from .audio import AudioChopperProfile
from .config import Config
from .connector import ConnectorSource
from csdr.csdr import dsp, output
from csdr import csdr
from csdr.pipe import Pipe
from datetime import datetime
from datetime import datetime, timedelta
from datetime import datetime, timezone
from datetime import datetime, timezone, timedelta
from .direct import DirectSource
from distutils.version import LooseVersion
from functools import partial
from functools import reduce
from glob import glob
from http.cookies import SimpleCookie
from http.server import BaseHTTPRequestHandler
from http.server import HTTPServer
from . import Controller
from . import SdrSource
from js8py.frames import Js8FrameHeartbeat, Js8FrameCompound
from js8py import Js8
from js8py import Js8Frame
from .map import Map, LocatorLocation
from .metrics import Metrics, CounterMetric
from multiprocessing.connection import Pipe, wait
from multiprocessing import Pipe
from operator import and_
from owrx.aprs import AprsParser
from owrx.audio import AudioChopper
from owrx.audio import AudioChopperProfile
from owrx.bands import Band
from owrx.bands import Bandplan
from owrx.bookmarks import Bookmarks
from owrx.client import ClientRegistry
from owrx.client import ClientRegistry, TooManyClientsException
from owrx.command import Argument, Flag
from owrx.command import CommandMapper
from owrx.command import CommandMapper, Flag, Option
from owrx.command import Flag
from owrx.command import Flag, Option
from owrx.command import Flag, Option, Argument
from owrx.command import Option
from owrx.config import Config
from owrx.connection import WebSocketMessageHandler
from owrx.controllers.api import ApiController
from owrx.controllers.assets import (
from owrx.controllers import Controller
from owrx.controllers.metrics import MetricsController
from owrx.controllers.session import SessionController
from owrx.controllers.settings import SettingsController, GeneralSettingsController, SdrSettingsController
from owrx.controllers.status import StatusController
from owrx.controllers.template import (
from owrx.controllers.websocket import WebSocketController
from owrx.cpu import CpuUsageThread
from owrx.details import ReceiverDetails
from owrx.dsp import DspManager
from owrx.feature import FeatureDetector
from owrx.feature import FeatureDetector, UnknownFeatureException
        from owrx.fft import SpectrumThread
from owrx.form import (
from owrx.http import RequestHandler
from owrx.js8 import Js8Parser
from owrx.js8 import Js8Profiles
from owrx.kiss import KissClient, DirewolfConfig
from owrx.kiss import KissDeframer
from owrx.locator import Locator
from owrx.__main__ import main
from owrx.map import Map
from owrx.map import Map, LatLngLocation
from owrx.map import Map, LocatorLocation
from owrx.meta import MetaParser
from owrx.metrics import Metrics
from owrx.metrics import Metrics, CounterMetric
from owrx.metrics import Metrics, CounterMetric, DirectMetric
from owrx.modes import Modes
from owrx.modes import Modes, DigitalMode
from owrx.parser import Parser
from owrx.pocsag import PocsagParser
from owrx.property import PropertyFilter
from owrx.property import PropertyLayer
from owrx.property import PropertyLayer, PropertyFilter
from owrx.property import PropertyLayer, PropertyStack
from owrx.property import PropertyManager, PropertyLayer
from owrx.property import PropertyStack
from owrx.property import PropertyStack, PropertyLayer
from owrx.pskreporter import PskReporter
from owrx.receiverid import ReceiverId
from owrx.sdr import SdrService
from owrx.service import Services
from owrx.socket import getAvailablePort
from owrx.source.connector import ConnectorSource
from owrx.source import SdrSource, SdrSourceEventClient
from owrx.source.resampler import Resampler
from owrx.source.soapy import SoapyConnectorSource
from owrx.users import UserList
from owrx.version import looseversion
from owrx.version import openwebrx_version
from owrx.websocket import WebSocketConnection
from owrx.wsjt import Ft8Profile, WsprProfile, Jt9Profile, Jt65Profile, Ft4Profile, Fst4Profile, Fst4wProfile
from owrx.wsjt import WsjtParser
from .parser import Parser
from .pskreporter import PskReporter
from queue import Queue, Full
from queue import Queue, Full, Empty
from .receiverid import ReceiverIdController
from .schedule import ServiceScheduler
from .session import SessionStorage
    from setuptools import find_namespace_packages
    from setuptools import PEP420PackageFinder
from setuptools import setup
from .soapy import SoapyConnectorSource
from socketserver import ThreadingMixIn
from string import Template
from subprocess import Popen
from .template import WebpageController
from unittest import TestCase
from unittest.mock import Mock
from urllib import parse
from urllib import request
from urllib.parse import parse_qs
from urllib.parse import quote
from urllib.parse import urlparse, parse_qs
from uuid import uuid4
import base64
import gzip
import hashlib
import hmac
import importlib.util
import inspect
import json
import logging
import math
import mimetypes
import os
import pkg_resources
import random
import re
import select
import shlex
import signal
import socket
import subprocess
import sys
import threading
import time
import wave
```

jketterl, 2021-01-09:
I'm afraid the main problem here is not the python dependencies. OpenWebRX relies on external software for the hardware interfacing and demodulation. Since those are written in C or C++, they would need to be made available on the OpenWRT platform somehow.

From what I remember, compiling software on the OpenWRT host itself may not even be possible (not enough memory?) so there's typically cross-compiling involved.

Also, I believe OpenWRT is available on a series of different CPU architectures, so making this available on OpenWRT probably would mean building the dependencies for any of these.

najdanovicivan, 2021-01-10:
Yes of course all the requred dependencies have to be compiled as well.

I’m gonna be using OpenWrt SDK fo x86_64 so there is no cross-compilations but there is cross compile toolchain when compiling on other architectures.

The way how openwrt works is that you just crate a makefile for the package. You can then compile the package on any architecutre you want

I first need to see what packages are already available on openwrt. And I’ll start porting the missing dependencies

It should be possible to all the dependencies as long as compilation with musl is possible

jketterl, 2021-01-10:
Compiling against musl should be possible, the result however will not be portable. This is due to the fact that both `csdr` and `owrx_connector` use vectorization for performance reasons. The resulting binaries will require certain cpu features (depending on the CPU that was used during the build). With glibc, this is mitigated by making use of FMV, but that functionality is missing in musl.

I was hitting that exact problem with the OpenWebRX docker images, they were initially based on Alpine. The solution was to switch to Debian base images.