jketterl/openwebrx #211: 8m band in bands.json

sq4bja, 2021-01-12:
Hi,
In several countries it is possible to use the 8m band. Could you please add 40 MHz band to bands.json:
   {
     "name": "8m",
     "lower_bound": 40025000,
     "upper_bound": 41,000,000,
     "frequencies": {
       "ft8": 40680000,
     }
   },
Recently, several European stations have appeared on FT8.

Tom SQ4BJA

legacycode, 2021-01-12:
Hi,
i could not find the 8m band in the official IARU band plans. Could you provide a document that shows the frequencies used?

* [IARU 1](https://www.iaru-r1.org/wp-content/uploads/2019/08/hf_r1_bandplan.pdf)
* [IARU 2](https://www.iaru.org/wp-content/uploads/2020/01/R2-Band-Plan-2016.pdf)
* [[IARU 3](https://www.iaru.org/wp-content/uploads/2020/01/R3-004-IARU-Region-3-Bandplan-rev.2.pdf)

On https://pskreporter.info/pskmap there are no stations that transmitted FT8 on 8m in the last 24h.

Best regards.

sq4bja, 2021-01-12:
There are no common regulations here and each country has a different approach to this band.
One of the regulations in Ireland: https://www.irts.ie/dnloads/Irish_8m_5m_band_plans.pdf
Also in South Africa, the band has been available since 2015:
http://ve7sl.blogspot.com/2015/04/new-band-for-zs-hams.html

Below you also have an interesting article about the 8m band:
https://ei7gl.blogspot.com/p/40-mhz.html

The range is very interesting. Today, via FT8, it was possible to hear the stations at distances of 900 and 1100 km from Slovenia and Croatia.

There are some stations on pskreporter. Just look:
[image]

sq4bja, 2021-01-12:
Of course, I can add 8m band manually, but when updating openwebrx the bans.json file will be overwritten.
I do not insist, but pay attention to the new band, on which the beacons are already working, and the FT8 emission takes place mainly on 40 680 MHz.

Tom SQ4BJA

jketterl, 2021-01-12:
I would like to prefer to keep the bandplan that gets installed with OpenWebRX at a more general level, that is to say it should contain the common, internationally applicable bands. That is already a challenge in its own, I'm pretty sure that it doesn't fulfill that expectations right now, especially with the VHF and UHF bands which tend to have greater variations from region to region due to their propagation habits (or lack thereof).

The bandplan is part of the configuration, and as such can be adopted pretty much freely by the operators, so the idea is that the bandplan that comes with OpenWebRX should be a good base for you to build on and easily adopted to your local regulations.

Contrary to my stance towards `config_webrx.py`, I wouldn't mind active exchange of bandplans. It might be a good idea to host them in the wiki for now. I have been thinking of including a bandplan editor into receiverbook, with the option to download a localized bandplan for use in OpenWebRX. Building an interactive editor is a lot of work, though, and I'm not sure it it would be accepted in the community.

As part of the configuration, the `bands.json` should not simply be overwritten by an OpenWebRX update. If the file has been edited, the package manager should give you the option to keep your file, or to overwrite it with the packaged one. Independently of your choice, the package manager should install the other option as `bands.json.dpkg-dist`, which gives you the option to merge both parties' changes using a diff tool of your choice.

jketterl, 2021-02-01:
Any final comments? I was expecting some more discussion after my last comment, but since that didn't happen I'd like to conclude that the default bandplan will remain as-is, and close this.

sq4bja, 2021-02-01:
Ok, sorry for not answering. You're probably right. Since there is no clear declaration, even in Europe, regarding the bandwidth, there is no point in adding it to the bandplan. However, will the FT8 station be heard on the 8m band? Will the color and legend for this band be properly assigned on the map?

Another solution could be to add directory support for example /etc/openwebrx/local containing individual settings that will not be overwritten by config_webrx.py and bans.json? Data not in the local directory will be retrieved from the correct files in the upper directory ../ .

Tom SQ4BJA

jketterl, 2021-02-01:
I don't understand why an FT8 station shouldn't be heard, the bandplan plays no role in this since you can decode FT8 on any frequency. If you have defined the band and it's frequencies correctly, you will see a green bookmark for the corresponding modes, it should work for background decoding, and it should be displayed correctly on the map.

As stated before: Once you have edited the `bands.json` file, the package manager will not overwrite your changes without asking you first. A file `bands.json.dpkg-dist` will be created containing the new bandplan from the package (assuming you chose to keep your local changes). This should be sufficient to be able to merge the changes from both sides.

The suggested solution to add a directory for local additions won't give you any notices of necessary changes to the files in this directory, so it is error-prone. Also, this solution would only allow additions, but no edits or removals.

sq4bja, 2021-02-01:
Ok, but if you add a new feature that isn't in my config_webrx.py, it won't be implemented anyway. It will appear only when I correct the configuration file. But if the basic data was in local, all new parameters would be taken from config_webrx.py along with the parameters recommended in the package.
Please take this as a user's suggestion only. A similar method of file management is used, for example, in SVXLink and works perfectly well.

jketterl, 2021-02-01:
You will always need to update your configurations when changes have been made. A local override directory will only work if the structure of the configuration remains unchanged, otherwise your local files will need to be updated as well.

I have worked with svxlink, I'm familiar with the system. I am however not convinced it is a good solution, especially not in svxlink where the local files contain functional code (tcl). It's also not really up to typical standards, normally you'd have a `conf.d` directory under `/etc`, whereas svxlink puts these files under `/usr/share`, which is not the correct location for configuration.

This is certainly not the last evolution, this is just how it works for now. As I said before, ideally we'd have a global database of bandplans from which you'd only pick the one that applies to your location (think something like the timezone selection). This might even be automated since the receiver location is available in the configuration. But there's a lot of work to be done until then.

closed 2021-02-01 by sq4bja