jketterl/openwebrx #212: <dev> FT8 and other decoding doesn't work

sq4bja, 2021-01-16:
Hi,
Decoding of digital broadcasts does not work after the last dev version update. Both in the background and in the user interface.
Openwebrx downloaded as a package from the Debian repository from the dev branch.

Tom SQ4BJA

jketterl, 2021-01-16:
First of all, I'd really appreciate if you could use the issue template.

Second, the log messages about the queue being full are probably just a secondary error, the real error should be further up in the log.

If I had to guess, I'd say you're probably missing some new config settings (`wsprnet_enabled`) that were introduced (happened to myself last night). I will make them optional.

sq4bja, 2021-01-16:
Hello, sorry for the wrong issue format.
You were right, the new configuration options have disabled ft8 decoding. After adding the missing options, everything started to work.

jketterl, 2021-01-16:
You can update the package from the repo now, too, I added a check and it should default wsprnet reporting to off if the config keys are missing.

Resolved?

sq4bja, 2021-01-16:
Yes, thank You!

closed 2021-01-16 by sq4bja