jketterl/openwebrx #215: Security: Arbitrary remote code execution vulnerability

dl9rdz, 2021-01-24:
**Describe the bug**
OpenWebRX is vulnerable to arbitrary remote code execution.

No problem if you use your own Openwebrx server on your isolated private network, but if you operate a server exposed to the public internet, you should be aware of the corresponding risks... (Anyone on the internet can run arbitrary programs on your openwebrx computer with the user id the server is running with (usually "openwebrx" in the debian default installation, the worst-case is if you manually installed it and run it as "root"); this most likely also implies full exposure of your local network)

**To Reproduce**
(details revealed to author by e-mail on Jan 1)

**Installation method**
Debian package / manual

**Versions**
0.20.1 (debian) / most recent git version from develop branch.

jketterl, 2021-01-24:
I'm sorry, but contrary to the text I did not receive a message containing any details. I just double-checked my inbox.

dl9rdz, 2021-01-24:
That's strange, I sent you the e-mail on Jan 1st, 17:09 from my address dl9rdz@darc.de to your address dd5jfk@darc.de. 

Sent it again right now...

jketterl, 2021-01-24:
Nope, still nothing, sorry. Checked all the spam folders, too. I'll send you another address to use.

jketterl, 2021-01-25:
Security issues have been addressed, the exploits provided have been successfully mitigated. Hotfix version 0.20.2 is currently being deployed to all repositories. Changes have been applied in `develop`, and latest development builds contain the fix as well.

jketterl, 2021-02-01:
@dl9rdz any more follow-ups on this? Am I missing E-Mails again? :wink:

I'd consider this closed, but I'd appreciate some final feedback.

dl9rdz, 2021-02-01:
Sorry for not responding earlier, I was kind'a busy last week.

Thanks for the super-fast fix after my mail finally got to you! I didn't do a thorough analysis of the update, only had a brief look, but seems to be a very solid fix.

jketterl, 2021-02-01:
Thank you :slightly_smiling_face:

If you do find some more after this, don't hesitate to open up a new issue. :slightly_smiling_face:

closed 2021-02-01 by jketterl