jketterl/openwebrx #217: Save squelch setting and default frequenchy per band in client cookie

RipperFox, 2021-02-20:
**Feature description**
Save the squelch setting in a cookie, so that when the clients reconnects (e.g. due to a server restart) or someone changes band on full receiver the squelch won't open.

Squelch setting could be set per band - also would be nice if the last used frequency per band would be saved.

**Target audience**
All users using squelch / single receivers that switch bands

jketterl, 2021-02-20:
First off: Cookies are a bad choice for this kind of interaction. They have been misused for client-side storage for years, completely ignoring the fact that they get sent to the server with every request. Luckily, HTML5 has introduced the localStorage API a few years ago, which is a much better solution.

Features are best described without going into the technical implementation. That part is better left to the developers.

Here's what's currently implemented:

The squelch can be configured per profile by the receiver operator already. This should in theory be sufficient for most use cases since wherever squelch typically makes sense (VHF, UHF and higher), the background noise level only shows little fluctuation. On the shortwave bands with the SSB modes, the squelch is typically left open.

The problem with this settings seems to be that many operators don't bother calibrating it, and there's no reasonable default either. If you encounter receivers where this is the case, I suggest contacting the receiver operator.

There is a very basic system that performs a similar task already by storing select parameters in the URL, which has been implemented by request before. It doesn't work well together with the profiles, but it has the advantage that these URLs can be shared with other users.

The squelch level is not carried over during profile changes since it typically doesn't apply to the next profile, even if a squelch setting would be useful on the new profile - the chances that the current level matches the new profile are pretty low. That means that on a profile change, you typically get the default squelch level for that profile - which is fully open if uncalibrated.

RipperFox, 2021-02-21:
About cookies: Completely agree - I'm no web dev, same data storage as client bookmarks was actually meant :)

The squelch can be configured or even selected per URL parameter, but all settings are completely temporary and gone as soon the server is restarted or someone switches profile. On my server you can e.g. choose 2m SSB, 2m FM, 70cm FM, etc. on a single RTL-SDR USB stick. Noise floor on VHF/UHF is usually very stable but I have to activate the squelch every time someone changes profile or the server restarts.

It would be convenient if start frequency, squelch level and squelch activation status could be client side stored per profile.
At least it would be nice to re-set the given squelch when reconnecting due to connectivity loss / server restart.

jketterl, 2021-02-21:
Did you configure `initial_squelch_level` for each of your profiles? To me, it sounds as if you did not. If so, please try that option first.

RipperFox, 2021-02-21:
Thank you very much, that worked flawlessly.

I diff'ed the standard conf of the deb last time there was a big change and adjusted my old config to new new format version and so totally missed this option - maybe worth including it in the example config?

jketterl, 2021-02-21:
Well, there's a lot of config options available... I wouldn't really want to explain every single one in the example config. That's what the documentation in the wiki is for.

I'm also working on a new web-based configuration interface right now, so I don't think that it's worth spending time on the configuration files. I hope that I can abandon them in a few releases.

closed 2023-03-17 by jketterl