jketterl/openwebrx #221: SDRplay API version 2 reference outdated

waterwin, 2021-03-01:
For API version 2, please use https://github.com/pothosware/SoapySDRPlay
For API version 3, please use https://github.com/SDRplay/SoapySDRPlay

Both links go to the same page for API version 3 probably because at the SDRplay website I can also only find the API 3 now.

It appears that API 2 is finally laid at rest.

jketterl, 2021-03-01:
I see... too bad, API version 3 hasn't seen any updates ever since it was still considered beta, and there's a still lot of problems with it...

I will remove Version 2 from the docs. It shouldn't be in use in any of the builds any more. Thanks for the hint.

waterwin, 2021-03-01:
Building with API 3 now on Dragonboard 410C ARM64.

closed 2021-03-01 by waterwin

jketterl, 2021-03-01:
good luck :slightly_smiling_face: 

wiki page has been updated and version 2 is no longer mentioned.

waterwin, 2021-03-01:
You ARE fast Sir.
;-)
Erwin

 
 

good luck 🙂