jketterl/openwebrx #223: WARNING: SoapySDR::Device::readStream timeout!

alexFab, 2021-03-06:
**Describe the bug**
waterfall in web-gui stops.

**SDR-Hardware**
USRP B210

**To Reproduce**
It happens after 10 - 60 min. At a sample rate of 4Msps and much earlier using  8Msps .

**Expected behavior**
A clear and concise description of what you expected to happen.

**Installation method**
- Ubuntu 19.10
- everything freshly complied from Source. 
  - https://github.com/pothosware/SoapySDR
  - https://github.com/pothosware/SoapyUHD
  - https://github.com/jketterl/owrx_connector

**Versions**
latest master branch

**Log messages**
WARNING: Soapy overflow
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!

**Additional context**
I'm not sure where the error is coming from. SoapySDR? SoapyUHD? owrx_connector? openwebrx? Or just some miss configuration?

thx a lot!

jketterl, 2021-03-06:
That's an error reported by the SoapySDR API, nothing OpenWebRX can handle. Please contact the developers of the SoapyUHD module.

closed 2021-03-06 by jketterl

n1zzo, 2021-03-07:
Hi @jketterl,
sorry for bringing this up again, but I believe that although the timeout is caused by the soapy library,
openwebrx is currently unable to recover after such an error.
Even in my issue: #219 , restarting soapyremote has no effect, and waterfall resumes only when openwebrx is restarted.
I believe that the issue should be solved by both sides, investigating the timeout cause, but also properly recovering from them on the openwebrx side.

jketterl, 2021-03-07:
Funny story, but the fact that the waterfall just stops without the device going into the "failed" state is because that already is a workaround that allows `airspyhf` devices to recover. They run into timeouts whenever certain parameters are changes. If that workaround wasn't in place, they would fail on profile switches.

Other than that, OpenWebRX is currently not designed to recover any device after a critical failure.