jketterl/openwebrx #224: Hackrf  is not work

BI4KIA, 2021-03-15:
I Installed kali system with PI 4，Plan to use Hackrf to run OpenWebRX v0.21.0-dev，compile item by item according to the tutorial，but it‘s’ not work.

OpenWebRX - Open Source SDR Web App for Everyone! | for license see LICENSE file in the package

Author contact info: Jakob Ketterl, DD5JFK dd5jfk@darc.de
Documentation: https://github.com/jketterl/openwebrx/wiki
Support and info: https://groups.io/g/openwebrx

2021-03-15 16:58:44,809 - owrx.main - INFO - OpenWebRX version v0.21.0-dev starting up...
2021-03-15 16:58:44,811 - owrx.config.migration - DEBUG - migrating config from version 1
2021-03-15 16:58:44,812 - owrx.config.migration - DEBUG - migrating config from version 5
2021-03-15 16:58:44,855 - owrx.sdr - ERROR - The SDR source type "hackrf" is not available. please check requirements.
2021-03-15 16:58:44,864 - owrx.sdr - ERROR - The SDR source type "rtl_sdr" is not available. please check requirements.
2021-03-15 16:58:44,864 - owrx.sdr - ERROR - The SDR source type "airspyhf" is not available. please check requirements.
2021-03-15 16:58:44,864 - owrx.sdr - ERROR - The SDR source type "sdrplay" is not available. please check requirements.
2021-03-15 16:58:44,865 - owrx.sdr - INFO - SDR sources loaded. Available SDRs:

SoapySDRUtil --probe="driver=hackrf"
######################################################

Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=hackrf
[INFO] Opening HackRF One #0 a063dc2c81800f...

-- Device identification
driver=HackRF
hardware=HackRF One
clock source=internal
part id=a000cb3c0055475c
serial=000000000000000000a063dc2c81800f
version=2015.07.2

-- Peripheral summary
Channels: 1 Rx, 1 Tx
Timestamps: NO
Other Settings:
* Antenna Bias - Antenna port power control.
[key=bias_tx, default=false, type=bool]

-- RX Channel 0
Full-duplex: NO
Supports AGC: NO
Stream formats: CS8, CS16, CF32, CF64
Native format: CS8 [full-scale=128]
Stream args:
* Buffer Count - Number of buffers per read.
[key=buffers, units=buffers, default=15, type=int]
Antennas: TX/RX
Full gain range: [0, 116] dB
LNA gain range: [0, 40] dB
AMP gain range: [0, 14] dB
VGA gain range: [0, 62] dB
Full freq range: [0, 7250] MHz
RF freq range: [0, 7250] MHz
Sample rates: 1, 2, 3, 4, 5, ..., 16, 17, 18, 19, 20 MSps
Filter bandwidths: 1.75, 2.5, 3.5, 5, 5.5, ..., 14, 15, 20, 24, 28 MHz

-- TX Channel 0
Full-duplex: NO
Supports AGC: NO
Stream formats: CS8, CS16, CF32, CF64
Native format: CS8 [full-scale=128]
Stream args:
* Buffer Count - Number of buffers per read.
[key=buffers, units=buffers, default=15, type=int]
Antennas: TX/RX
Full gain range: [0, 61] dB
VGA gain range: [0, 47] dB
AMP gain range: [0, 14] dB
Full freq range: [0, 7250] MHz
RF freq range: [0, 7250] MHz
Sample rates: 1, 2, 3, 4, 5, ..., 16, 17, 18, 19, 20 MSps
Filter bandwidths: 1.75, 2.5, 3.5, 5, 5.5, ..., 14, 15, 20, 24, 28 MHz

jketterl, 2021-03-15:
Please use the issue template.

> 2021-03-15 16:58:44,855 - owrx.sdr - ERROR - The SDR source type "hackrf" is not available. please check requirements.

Please check the feature report, HackRF section.

https://github.com/jketterl/openwebrx/wiki/API#httpyour-urlip8073apifeatures

jketterl, 2021-03-18:
No response, so assuming this is not actually a bug. Closing.

Please use the mailing list for further support.

closed 2021-03-18 by jketterl