jketterl/openwebrx #225: Incorrect range setting.

IkaSGC, 2021-03-23:
Hi, I'm completely new to this WebSDR. I have the basic settings here "RTL-SDR USB Stick" for 2m and 70cm. 

But my SDR handles 100khz - 1.7 GHz. How can I reset these basic settings? Is the correct configuration file available somewhere? Thank you.

[image]
[image]

jketterl, 2021-03-23:
This is not an OpenWebRX issue. Please use the mailing list for support requests.

closed 2021-03-23 by jketterl

IkaSGC, 2021-03-23:
Where i find that mailing list??

jketterl, 2021-03-23:
You went right past it: When you click "New issue", there's three options. The third one is the mailing list.

https://groups.io/g/openwebrx