jketterl/openwebrx #226: openwebrx.de - image download and configuration

om2abc, 2021-03-27:
Dear firends, I downloaded image from https://www.openwebrx.de/download/rpi.php. Image is fully functioning and preset to several SDR receivers. However, my problem is that I don't know how to change the settings, I can't get to the main root. Please do you advise what commands I need to enter to change header, frequencies and other settings? Thank you, 73 Stefan - OM2ABC

jketterl, 2021-03-27:
This is not an OpenWebRX issue. Please use the mailing list for support.

Besides, there is plenty of documentation in the Wiki: https://github.com/jketterl/openwebrx/wiki/Configuration-guide

closed 2021-03-27 by jketterl

om2abc, 2021-03-27:
> Odoslané: sobota 27. marca 2021 18:54
> Komu: jketterl/openwebrx ***@***.***>
> Kópia: om2abc ***@***.***>; Author ***@***.***>

Thank you Jakob for instruction, but on recomanded page is missing how I change the base setup from image?
After installation of image I have/I see just command lines. Could you help me please and send me instruction whit which command I need to use for base configuration setup?

Many thansk for your help

73! All the best to you

Stefan - OM2ABC

--- -- ..--- .- -... -.-.

Tel.: +421 903 70 50 70

E-mail: ***@***.***