jketterl/openwebrx #227: Unable to decode APRS in development branch

ofadam, 2021-04-18:
**Describe the bug**
No longer able to decode packet (APRS) in development branch.

**To Reproduce**
Start Openwebrx from fresh boot, running as service.
I see no error messages in the running log related to Direwolf.

**Expected behavior**
When on APRS frequency (144.39 in US) and with packet mode enabled, it normally decodes in overlay box. Audio is usually muted by default as well (this is no longer the case.)

**Installation method**
Originally installed stable Ubuntu repo, but switched to development branch for great web settings interface.

**Versions**
Using dev: 04a5e6705fb3fd42dcf3cc9f1ab6d6362e33b9df

jketterl, 2021-04-18:
I have no problems decoding APRS here, both in the live receiver and as a background service.

This is odd though:

> Audio is usually muted by default as well (this is no longer the case.)

Audio should not be muted in packet mode, and this hasn't been any different in the past.

Is this maybe in relation to using packet as `start_mod` or in a "regular" (i.e. yellow) bookmark?

ofadam, 2021-04-18:
Ah, my apologies - I apparently forgot that you could hear packet by default.

My issue may have something to do with this bit of log I finally was able to spot? Since this is only affecting me, I can move this to the support forum if need be.

Apr 18 14:57:52 BasementServer openwebrx[35268]: 2021-04-18 14:57:52,918 - owrx.audio.queue - ERROR - failed to decode >
Apr 18 14:57:52 BasementServer openwebrx[35268]: Traceback (most recent call last):
Apr 18 14:57:52 BasementServer openwebrx[35268]:   File "/usr/lib/python3/dist-packages/owrx/audio/queue.py", line 70, >
Apr 18 14:57:52 BasementServer openwebrx[35268]:     job.run()
Apr 18 14:57:52 BasementServer openwebrx[35268]:   File "/usr/lib/python3/dist-packages/owrx/audio/queue.py", line 41, >
Apr 18 14:57:52 BasementServer openwebrx[35268]:     raise RuntimeError("decoder return code: {0}".format(rc))
Apr 18 14:57:52 BasementServer openwebrx[35268]: RuntimeError: decoder return code: 2

jketterl, 2021-04-18:
No, that looks more like an error from a WSJT-X mode, the audio queue is used for the "chopped" audio samples there.

This may be an incompatibility with the specific Ubuntu environment. The packages in the stable repository are still targeted at Focal Fossa, but the experimental repository has been updated to Groovy Gorilla (and it looks like that I will have to upgrade it to Hirsute Hippo soon). I will need to set up a testing environment to try it out.

Is there a way you can get me a full log from the start of the connection until the point where you'd expect it to decode APRS, please?

ofadam, 2021-04-18:
I think I've discovered that it was a combo of two different issues on my end:

- Although I normally run openwebrx as a service under a dedicated system user, I was troubleshooting it from the bash prompt while root. That created some /tmp files that apparently weren't able to be overwritten by the openwebrx user (clearing them fixed those errors.)

- I've been using RTL-SDR using auto gain, but noticed a log entry saying that the audio level is too high for Direwolf. I set the gain to 30 instead and it is now decoding as expected!

ofadam, 2021-04-18:
Being able to change gain settings without restarting openwebrx via web settings is just a beautiful thing - I thank you.

jketterl, 2021-04-18:
Yeah I spent a lot of time in the past weeks making sure that as many settings as possible can be changed on the fly in the web config, so feel free to experiment.

I'm having a hard time choosing one of the problem options... The files in `/tmp` shouldn't be that bad since OpenWebRX typically generates new filenames, it uses the memory locations of the objects to produce filenames that are guaranteed to be unique during runtime. There's nothing to guarantee for them to be unique across restarts, but the system is typically random enough to avoid conflicts.

The AGC may cause problems in combination with other strong signals on the input, since that has the potential to distort the signal of interest.

Either way, I'm glad it's working for you now :)

ofadam, 2021-04-18:
My best guess on the temp files is that there are a few without timestamped filenames that may have been throwing errors in the log that weren't actually applicable to the APRS problem. But that's just a hunch.

I appreciate all the help today - I'll go ahead and close this, as everything is working nicely.

closed 2021-04-18 by ofadam