jketterl/openwebrx #228: Unable to edit either Device or Profile names in web settings

ofadam, 2021-04-23:
**Describe the bug**
No changes made to existing device or profile names appear on the websdr main screen.

**To Reproduce**
In the web settings, try to change existing device or profile name. Click "apply and save", then reload websdr.

**Expected behavior**
Changing the names should change what appears in the websdr drop-down menu options, but it does not.

**Installation method**
Development branch of Ubuntu repo

jketterl, 2021-04-23:
Referring to #227: Please check the permissions in `/var/lib/openwebrx`

ofadam, 2021-04-23:
Everything seems okay? I opened settings.json and it does indeed reflect the changes I made. I thought it might be a browser cache issue, but I get the same old names when opening a fresh browser just installed that has never visited this websdr.

-rw-r--r--  1 openwebrx openwebrx  3787 Apr 23 22:06 bookmarks.json
-rw-r--r--  1 openwebrx openwebrx 14515 Apr 23 22:40 settings.json
-rw-------  1 openwebrx openwebrx   757 Apr 18 19:15 users.json

jketterl, 2021-04-23:
Yeah, permissions look OK. Not sure what's happening, I have spent a lot of time to make the system propagate changes to any running clients immediately. Any errors in the log? Did you add devices or profiles in this session?

ofadam, 2021-04-23:
I did add profiles in this session, then tried rename a few afterward. Those changes took in the settings.json file, though, but don't appear in the main web interface. No errors in the log. Other changes to profile like initial frequency and gain DO take effect as expected. It's just the name that does not!

jketterl, 2021-04-23:
alright... if it's just the name, i have a suspicion where the error might be, but i don't know what kind of error to look for.

does this problem still exist after a restart?

ofadam, 2021-04-23:
After restarting openwebrx, all names are correct!

jketterl, 2021-04-24:
Alright, turns out the names were not wired at all. I don't know how this even remotely passed my testing, but I also cleared some other event-related stuff lately, so maybe this was covered by sheer luck previously. It wasn't easy to hook up the events the right way since I had to cover all the potential source and profile add and remove events as well, but it seems to work now.

ofadam, 2021-04-25:
Works like a charm! Thank you!

closed 2021-04-25 by ofadam