jketterl/openwebrx #230: No waterfall window for WSJT-X mode decodes

ofadam, 2021-04-25:
**Describe the bug**
When selecting a digital mode to decode that uses WSJT-X (FT8, WSPR, etc), the smaller waterfall display that once showed is now just red.

**To Reproduce**
Select FT8 frequency, make sure decoding FT8.

**Expected behavior**
A zoomed-in version of the waterfall once displayed where you could see the transmission.

**Installation method**
Ubuntu package for bleeding edge

**Log messages**
No errors in the log at all.

[image]

jketterl, 2021-04-25:
That looks like what you typically get with any digimode and the squelch closed. What's the squelch setting?

ofadam, 2021-04-26:
Squelch is set for the band via web settings, but otherwise untouched – it appears to be grayed out as a button/slider, but does seem to be activating despite the audio being uninterrupted.

jketterl, 2021-04-26:
The squelch should not be "grayed out" in FT8. The secondary waterfall sits after the squelch, so if the squelch is closed, you don't get a secondary waterfall. Can you open the squelch?

ofadam, 2021-04-26:
I must have been tired, because my original ticket no longer makes sense. It appears that selecting a digital mode doesn't automatically disable squelch like I remember it doing before. If I manually "turn down" squelch, everything works as expected. But that wouldn't help background decoding?

jketterl, 2021-04-26:
It depends on the mode. Some have the squelch disabled, e.g. APRS.

One could argue that squelch doesn't make sense for FT8, or even SSB in general, but I haven't done any restrictions here (yet?).

Background decoding always runs with the squelch fully open. It doesn't even look at the `initial_squelch_level` setting.

ofadam, 2021-04-26:
Good to know on background decoding being unaffected. Just my personal preference, but I'd be game for squelch being disabled for all digital modes at minimum. Default muting for FT8 et all is my personal preference as well, but I may rely on the visual queues for signals being present and others may prefer hearing some audio?

jketterl, 2021-04-26:
Muting is a different beast since the underlying mode is still USB, which I wouldn't want to mute. I don't want to make things extra complicated by adding even more conditions, and as you already said, this is personal preference. I'd say that's what the mute button is for.

As for the squelch, I still have no idea on how to get a proper quorum for decisions like that, and I don't want to change it without. Personally, I don't care too much since the overlap between bands where you'd setup squelch and bands where you'd use FT8 is not really big.

jketterl, 2021-04-29:
so, what should we do with this ticket now?

closed 2021-05-06 by jketterl