jketterl/openwebrx #232: release-1.0: ModuleNotFoundError: No module named 'owrx.form.input'

sgrams, 2021-05-09:
**Describe the bug**
Running a fresh clone of release-1.0 results in a following error:

```
Traceback (most recent call last):
  File "/home/sjg/git/openwebrx/./openwebrx.py", line 3, in <module>
    from owrx.__main__ import main
  File "/home/sjg/git/openwebrx/owrx/__main__.py", line 8, in <module>
    from owrx.http import RequestHandler
  File "/home/sjg/git/openwebrx/owrx/http.py", line 1, in <module>
    from owrx.controllers.status import StatusController
  File "/home/sjg/git/openwebrx/owrx/controllers/status.py", line 3, in <module>
    from owrx.sdr import SdrService
  File "/home/sjg/git/openwebrx/owrx/sdr.py", line 4, in <module>
    from owrx.source import SdrSource, SdrSourceEventClient
  File "/home/sjg/git/openwebrx/owrx/source/__init__.py", line 15, in <module>
    from owrx.form.input import Input, TextInput, NumberInput, CheckboxInput, ModesInput, ExponentialInput
ModuleNotFoundError: No module named 'owrx.form.input'
```

**To Reproduce**
Steps to reproduce the behavior.

git clone -b master https://github.com/jketterl/openwebrx.git
cd openwebrx
./openwebrx.py

**Expected behavior**
No error should be seen...

**Installation method**
How did you install OpenWebRX? (Raspberry Pi SD card image, Debian / Ubuntu packages, Docker image, manually?)
Manual installation.

**Versions**
What version of OpenWebRX are you running? (Check on startup, or see `owrx/version.py`. If a `-dev` version is used, ideally state the commit the issue is appearing on)
release-1.0

**Log messages**
Are there any relevant messages relating to the bug in the output / log of OpenWebRX? (On most installations, the log should be available using the command `sudo journalctl -u openwebrx`)
As in the description.

**Additional context**
Add any other context about the problem here.

Reverting back to previous release works fine.

sgrams, 2021-05-10:
Sure,

```
mainframe :: ~/git/openwebrx » git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

```
mainframe :: ~/git/openwebrx » find . -name "__pycache__"
./owrx/audio/__pycache__
./owrx/controllers/__pycache__
./owrx/__pycache__
./owrx/source/__pycache__
./owrx/config/__pycache__
./owrx/property/__pycache__
```
After removing all the __pycache__ subdirectories the issue still exists.

jketterl, 2021-05-10:
Alright then... what's your base system and python version, please?

sgrams, 2021-05-10:
Arch Linux, kernel as below:
```
mainframe :: ~ » uname -a
Linux mainframe 5.11.16-arch1-1 #1 SMP PREEMPT Wed, 21 Apr 2021 17:22:13 +0000 x86_64 GNU/Linux
```
```
mainframe :: ~ 130 » python --version
Python 3.9.4
```
Please note, that reverting it back to the previous release works just fine. I wouldn't blame it on the system configuration.

jketterl, 2021-05-11:
That's more than just weird. PEP420 (since python 3.3) should allow a module to exist without `__init__.py`, so that example is not necessarily the only way to build packages. That PEP is more than confusing though, and I've had issues with it before. What makes this even more odd is that many python installations will indeed run this code without any issues...

I can throw the file in there since it doesn't hurt anybody, but I'd really like to understand what is going on here just to avoid it in the future.

jketterl, 2021-08-04:
closing since this should be fixed with 1.1.

closed 2021-08-04 by jketterl