jketterl/openwebrx #234: CookieError when there are other web services in the IP

ra1nb0w, 2021-05-10:
**Describe the bug**
the web application faults when there are other cookies for the same hostname.

**To Reproduce**
connect to the openwebrx web interface after used another web service (like, tvheadend) on the same hostname.

**Expected behavior**
openwebrx web interface

**Installation method**
through Debian packages (stable)

**Versions**
1.0.0 but tested also with latest experimental (1.1.0)

**Log messages**
```
May 10 18:56:37 hos openwebrx[4575]: Exception happened during processing of request from ('192.168.178.203', 56994)
May 10 18:56:37 hos openwebrx[4575]: Traceback (most recent call last):
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
May 10 18:56:37 hos openwebrx[4575]:     self.finish_request(request, client_address)
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
May 10 18:56:37 hos openwebrx[4575]:     self.RequestHandlerClass(request, client_address, self)
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3/dist-packages/owrx/http.py", line 40, in __init__
May 10 18:56:37 hos openwebrx[4575]:     super().__init__(request, client_address, server)
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
May 10 18:56:37 hos openwebrx[4575]:     self.handle()
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/http/server.py", line 426, in handle
May 10 18:56:37 hos openwebrx[4575]:     self.handle_one_request()
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/http/server.py", line 414, in handle_one_request
May 10 18:56:37 hos openwebrx[4575]:     method()
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3/dist-packages/owrx/http.py", line 46, in do_GET
May 10 18:56:37 hos openwebrx[4575]:     self.router.route(self, self._build_request("GET"))
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3/dist-packages/owrx/http.py", line 55, in _build_request
May 10 18:56:37 hos openwebrx[4575]:     return Request(self.path, method, self.headers)
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3/dist-packages/owrx/http.py", line 68, in __init__
May 10 18:56:37 hos openwebrx[4575]:     self.cookies.load(headers["Cookie"])
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/http/cookies.py", line 529, in load
May 10 18:56:37 hos openwebrx[4575]:     self.__parse_string(rawdata)
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/http/cookies.py", line 593, in __parse_string
May 10 18:56:37 hos openwebrx[4575]:     self.__set(key, rval, cval)
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/http/cookies.py", line 485, in __set
May 10 18:56:37 hos openwebrx[4575]:     M.set(key, real_value, coded_value)
May 10 18:56:37 hos openwebrx[4575]:   File "/usr/lib/python3.7/http/cookies.py", line 352, in set
May 10 18:56:37 hos openwebrx[4575]:     raise CookieError('Illegal key %r' % (key,))
May 10 18:56:37 hos openwebrx[4575]: http.cookies.CookieError: Illegal key 'ys-api/mpegts/service'
```

**Additional context**
same problem if using nginx reverse proxy or directly connection.

jketterl, 2021-05-10:
That's an error somewhere between the cookie parser included in python and the cookies set by tvheadend. I don't really know which of them is in the wrong, but I don't have any influence over neither of them, so this is nothing that can be fixed here.

closed 2021-05-10 by jketterl

ra1nb0w, 2021-05-10:
Just for reference: https://bugs.python.org/issue44104
Anyway, seems an old problem already discussed (2008).

ra1nb0w, 2021-05-11:
The cookie (python code) can be an array or a single string where the single cookies are separated by semicolon and the variable from data by equal sign. With this, you can use a regexp, split and check if there is a slash in the variable name.

ra1nb0w, 2021-05-13:
Just to conclude this: https://bugs.python.org/issue44104#msg393539

jketterl, 2021-05-13:
Now I'm wondering how one filters out `CookieError`s. Filtering Exceptions / Errors doesn't really make sense to me in this situation... I suppose the underlying assumption is that we're feeding cookies one by one, and that the expectation is that we should filter out the ones where errors occur... What am I missing here?

There's a lot of logic in their `__parse_string()` method that I wouldn't want to duplicate... and that's where the loop takes place where we could filter. If the API would return `parsed_items`, it would be possible, but I don't see how that would work...

ra1nb0w, 2021-05-13:
Do you mean something:
```
if "Cookie" in headers:
            try:
                self.cookies.load(headers["Cookie"])
            except CookieError as exc:
                raise (...somthing...)
```
with filtering per cookie?

jketterl, 2021-05-13:
more like

```
parsed_items = self.cookies.parse_string(headers["Cookie"])
for item in parsed_items:
    try:
        self.cookies.load(item)
    except CookieError:
        pass
```

won't work, of course

ra1nb0w, 2021-05-14:
ok. something like this:
```
if isinstance(headers["Cookie"], str)
   cookies = headers["Cookie"].split(';')
   for item in cookies:
       try:
           self.cookies.load(item)
       except CookieError:
           pass
```

jketterl, 2021-05-14:
No, absolutely not. Checkout the complexity in https://github.com/python/cpython/blob/main/Lib/http/cookies.py#L539 - there's way more parsing than just `.split(';')`.

ra1nb0w, 2021-05-14:
Yes, I read but that will be done at `.load()` for validity and security. You essentially need only to split the string (split the parsing) not to parse them (best with `.split('; ')`). Anyway, I asked upstream for an example.

jketterl, 2021-05-14:
I know that. But I don't want that because that functionality should be covered by the library. Also, I don't think that simply splitting and loading the individual parts is fully equivalent to what the library does, even if the individual parts still go through the same validation. If you look closely, there is no `.split('; ')` in the library, but instead a pretty complex regex. 

(Looking at that part again makes me wonder how the invalid characters even made it through the regex...)

The point I'm trying to make here is that if I'm supposed to filter the individual cookies, the library should be able to expose that functionality in separate calls so that I'm able to do it. Basically take `.__parse_string()` and split it into two parts, since it already has the parsing part and the loop portion at the bottom where filtering could take place.

ra1nb0w, 2021-05-16:
Clearly, it is not a splitter but a parser (as I said).
I got the point, maybe they don't think it is necessary or it is not so easy to discover how to do that. Anyway, no more answer from the python issue.