jketterl/openwebrx #237: edit button bookmarks

cptechnik, 2021-05-13:
**Describe the bug**
double click is not possible at mobile browser (android)

**To Reproduce**
double click on bookmark entry 

**Expected behavior**
extra edit button to activate the edit form 

**Versions**
newest on repository

jketterl, 2021-05-13:
I'm sorry, but as OpenWebRX is generally not very mobile friendly, I didn't expect anybody to use the web config on mobile either, so this has never even been tested on mobile. The need to fix this problem really comes down to how many people actually need or want to edit bookmarks on a mobile device...?

cptechnik, 2021-05-24:
me and one guy behind me... 8-)

Just change the double-click into a edit-button...
"thats all"  8-)
...its not really urgent...

import and export were also nice... but also not urgent...