jketterl/openwebrx #238: use openstreetmap instead of gmap

cptechnik, 2021-05-13:
I wish that it is possible to use openstreetmap instead of google maps...
...because its free...!

jketterl, 2021-05-13:
Sorry, I've said repeatedly that I am not going to maintain any additional map providers. My personal preference is Google Maps because it works, so that's what I am going to provide and maintain myself. If you want others, you are free to contribute, but I will require you to commit to maintaining these parts in the future.

closed 2021-05-13 by jketterl

cptechnik, 2021-05-23:
http://www.comapp-online.de/osm3/
OSM-Generator
...wenn man will...!?
...if sombody **wants** to change...

jketterl, 2021-05-23:
yeah but you don't understand: you're the one that wants.

cptechnik, 2021-05-23:
No, i am the one who has a gh account, speak a little bit English and post something... and stands for his opinion...

jketterl, 2021-05-23:
An issue is more than just an opinion, it is a request. There is other, more suitable places to express opinions. As long as you don't intend to put some effort behind your opinion, this is a fruitless and unnecessary discussion since an opinion has little effect here.

cptechnik, 2021-05-23:
I have a few guys behind me, but they don't speak English as I, and they are happy how to use a smaatfohn, and by the way they often use my sdr by mobile devices, and another had installed owrx on rpi and configurate it by mobile device...
But they never be able to log on GitHub, and I am not be able to fit osm in to the code... I am just to begin to understand how GitHub works...
If you don't hear somebody, it's not only because they are not there... sometimes it's just they don't able to speak...
But you explained your situation, 
you don't want to build it.

jketterl, 2021-05-23:
I hear a lot of people that would like to see OSM. This is not the first request. In fact, there has already been a half-OSM solution in place (https://github.com/jketterl/openwebrx/pull/72), but since it stopped working, and nobody fixed it, I removed it. I don't see any reason to invest my time into a second map implementation as long as the current map works (and it does).

Your request for OSM "because it's free" implies that Google Maps costs money, which is not immediately true. Google does provide every user with a very generous free tier. I'm running multiple receivers plus receiverbook off mine, and I have yet to pay my first Dollar. The only restriction Google has made is that they are requiring your credit card information before you can start using it because too many freeloader were skipping the bill in the past.

I don't have a problem with the ideology behind OSM, but the quality of their software is not on the same level as that of Google Maps. That is the sole reason why the current map is a Google Map.

This is open source, so nobody is preventing you from building your own map with OSM. If you commit to maintaining it in the future, I would even consider merging it into mainline. If you really believe in OSM and want to support them, this is your chance. But please don't expect me to do it just because you think it's better because "it's free".

So, there you have it, that's my opinion.

cptechnik, 2021-05-24:
>> "because it's free" implies that Google Maps costs money,
Not at all... its free means, no google can change the cost, the rights to use, its free to use... google is free of costs at the moment, and if "you" use a key, and... other rights you have to accept... 
This is the main thing... not only the costs...

> Und OpenStreetMap? Der Anspruch hinter dem Projekt OpenStreetMap ist kein geringerer als eine Karte der Welt zu erstellen, die weder Google noch der NASA gehört, sondern - wie Wikipedia - von Allen erstellt wird und Allen gehört.   >>>> http://www.mediensyndikat.de/30_20.html

>> quality of their software
this is the problem... Google invents more money on this, than free programmers can do, or organizations who spends money...
but, ... WE could use, not because its free of charge, because its free of use... and often exacter than google... and when not... WE can change the map... 

I am not fit in js, i learnd on my own html and php... a little bit, i would say...

This is a simple code to implement a map of OSM.

`<!DOCTYPE html>
<html>
<head>
	<meta charset="utf-8" />
	<title>comapp openstreetmap-Generator</title>
	<link rel="stylesheet" type="text/css" href="http://www.comapp-online.de/osm3/shared/css/generator_3.0/map.css" />
</head>
<body id="fe">
	<div id="map"></div>
	<script type="text/javascript" src="http://openlayers.org/api/OpenLayers.js"></script>
	<script type="text/javascript" src="http://www.openstreetmap.org/openlayers/OpenStreetMap.js"></script>
	<script type="text/javascript" src="http://www.comapp-online.de/osm3/shared/js/jquery_1.8.3/jquery.min.js"></script>
	<script type="text/javascript" src="http://www.comapp-online.de/osm3/shared/js/plugins/jquery.flash.js"></script>
	<script type="text/javascript" src="http://www.comapp-online.de/osm3/shared/js/plugins/modernizr_2.6.2.js"></script>
	<script type="text/javascript" src="http://www.comapp-online.de/osm3/shared/js/generator_3.0/mapGenerator.min.js"></script>
	<script type="text/javascript">
		(function($){
			// ********** - comapp - mapGenerator jsonData - start - ************ //
			var json = {"version":3,"locale":"de","units":"m","flashPath":"http://www.comapp-online.de/osm3/shared/swf/audioPlayer.swf","iconPath":"http://www.comapp-online.de/osm3/shared/img/marker/","geoPath":"http://www.comapp-online.de/osm3/shared/ajax/geopos.php","title":"Mein Titel","descriptionStartStatus":1,"descriptionText":"Mein Beschreibungstext","descriptionButtonShowText":"Informationen und Hilfe zur Karte anzeigen","descriptionButtonHideText":"Informationen und Hilfe zur Karte verstecken","audioPlayImage":"http://www.comapp-online.de/osm3/shared/img/play.png","audioStopImage":"http://www.comapp-online.de/osm3/shared/img/stop.png","audioDownloadImage":"http://www.comapp-online.de/osm3/shared/img/download.png","geoDownloadImage":"http://www.comapp-online.de/osm3/shared/img/geo.png","lon":11.7,"lat":49.6,"zoom":5,"panzoombar":1,"permalinkchk":1,"scaleline":1,"showPopupOnHover":true,"markers":[{"markerName":"Mein neuer Marker","markerLon":"7.36","markerLat":"50.49","markerIconImage":0,"markerContent":"Meine neue Beschreibung des Markers"},{"markerName":"Mein neuer Marker","markerLon":11.7,"markerLat":49.6,"markerIconImage":0,"markerContent":"Meine neue Beschreibung des Markers"}],"requiredField":"Dieses Feld muss ausgefüllt werden."};
			// ********** - comapp - mapGenerator jsonData - end - ************ //
			$.fn.getJsonData = function() {
				return json;
			};
			$('body').mapGeneratorModel('initVal');
			$('body').mapGeneratorMap('buildMap');
			$('body').mapGeneratorMap('setBaseMode');
		})(window.jQuery);
		
		
		if (!window.console) console = {
			log: function() {},
			info: function() {},
			warn: function() {},
			error: function() {},
			debug: function() {}
		};
		function setFlashStatus() {
			$('body').mapGeneratorModel('setFlashStatus',true);
		}
	</script>
	<script type="text/javascript" src="http://mediensyndikat.de/geo/generator2.js"></script>
</body>`

I can build a html around... and change with php the vars of js code... but... 

...and i would generate several sample codes to watch which code is for this and that additional function...

The creator can also create forms...
https://wiki.openstreetmap.org/wiki/DE:Markermap
http://www.mediensyndikat.de/generator/

cptechnik, 2021-05-24:
I found
https://github.com/jketterl/openwebrx/blob/develop/htdocs/map.html
https://github.com/jketterl/openwebrx/blob/develop/htdocs/map.js
https://github.com/jketterl/openwebrx/blob/develop/htdocs/lib/settings/MapInput.js
is this all map related?

jketterl, 2021-06-02:
You do notice the irony in quoting the openness of OSM while I've been emphasizing that this project is just as open and you are free to add OSM right from the start? Are you serious?

You seem to have identified the right files... The first two in the list are used for the user-facing map, while the third one is the code that provides the map to select the receiver location in the web config. It uses an [this](https://github.com/cyphercodes/location-picker) module.

You will need to find a different way of communication if you wish to ask further questions though. I do not intend to give developper support via a closed github issue.