jketterl/openwebrx #239: Deafult Log and Status dispays off when OWR is opened

G8JNJ, 2021-05-14:
Currently when a user opens OWR, the Log and Status displays are initially shown on top of the waterfall.

Most users do not need to see these items unless they are trying to debug a problem, so I would suggest that the default opening condition should be that they are not displayed until they are selected with the appropriate Icon on at the top of the web GUI.

This would provide a cleaner looking GUI to a new user, whilst still retaining to option to select it if required.

jketterl, 2021-05-14:
I'll have to change this into an idea. This basically just expresses an opinion, but before I actually change this, I'd like to see at least some kind of discussion and consent in the community. It's not actionable in the meantime.

jketterl, 2022-07-27:
Please notice: A poll has been initiated on the groups.io mailing list. Please express your opinion there: https://groups.io/g/openwebrx/topic/92650248