jketterl/openwebrx #240: OWR 1.0 provides a new settings Icon to open an admin panel - option required to remove it from GUI

G8JNJ, 2021-05-14:
OWR 1.0 provides a new settings Icon to open a new admin panel which requires a user name and password to open.

This feature is intended for use only by admins and is not needed by casual users.

Would it be possible to provide an option to remove this feature from the web GUI and only provide access to the admin page by appending the OWR url with /admin ?

As it is primarily an admin function it's availability to casual users should be obscured.

This would avoid confusion as to it's purpose for casual users, who may otherwise wonder what it is for and try to access it.

jketterl, 2021-05-14:
Unfortunately, I cannot identify admins other than by them logging in. That means that there needs to be some kind of UI element for them to initiate the login, and it cannot be conditional.

I don't believe in security by obscurity either, so that argument is void for me.

closed 2021-05-14 by jketterl

G8JNJ, 2021-05-14:
I agree the admin function still needs a login and teh admin page access should be the same as it is now. I'm just suggesting the option to hide the settings button and provide access just via the url string, as the Icon is currently placed among the user functions when it is not really for use bay anyone other than the admin.

The purpose is to make the user GUI less complicated.

jketterl, 2021-05-14:
Yeah I understand, but I'm not going to face the amount of support requests of those who won't find it then.

G8JNJ, 2021-05-14:
Hi Jakob,
Understood about hiding the Settings Icon from the web GUI, but I bet the admins will get almost as many questions from casual users about what they can't open the settings menu with out a user name and password and how can they go about obtaining one.
I wasn't suggesting security via obscuring the option, it was more about not confusing casual users about the purpose of the settings menu, which they may believe to be another user option as is the case with the Status, Log, Receiver and Map icons that are placed along side of it.

If an admin can't understand how to access the setup menu by adding /settings to the end of the url, then they are definitely going to struggle with the rest of the configuration. 

Sorry, I thought I had been correctly using the template you provided in order to request new features.
I previously got told to use this method to request new features rather than to post them on the IO group. 

But are you now suggesting that it would be better to post suggestions onto the group first for discussion purposes and then place a request on Github if the response is favourable ? 

I'm more than happy to comply with whatever method suits you the best, but I'm currently not sure if I'm doing things correctly.

Thanks again for all you hard work and consideration.
Regards,
Martin

jketterl, 2021-05-14:
As for the issue at hand: I'm not aware of such requests or complaints. If you do have a suggestion about a different UI solution, feel free to open a new issue, but as I said, the current suggestion of simply removing the icon does not work out for me.

As for your questions concerning the issues: The templates are there to get some very basic answers right away that I would have to ask anyway as the issue progresses. I don't see the structure of the template in your tickets, so I was assuming you didn't use them. There's enough information for me not to worry too much about it. Still, if you request multiple features, there should be multiple issues, just to separate the corresponding discussion. Also, if I finish a feature, I'd like to be able to close the corresponding ticket.

I have been very strict about what kind of discussion I am having on here: This space is reserved for bugs and feature requests only. Anything else should go to the mailing list, but I'm basically allowing bugs and feature requests there, too. Your feature requests are valid on here, but some of them just don't work out (I would have given you the same answers if you had posted them to the mailing list). The bookmarks issue contains three separate things (bookmark notes, bookmark colors, bookmark imports) that are better implemented and discussed separately.

G8JNJ, 2021-05-14:
Hi Jakob,
OK understood.
I'll try and use the appropriate method as required.

Regards,
Martin