jketterl/openwebrx #241: OWR provides a new bookmark editor - request for notes field and Import / Export and Sort

G8JNJ, 2021-05-14:
OWR provides a new bookmark editor.

This permits easy editing and the addition of new frequency markers.

Would it be possible to add a new 'Notes' field, which could contain more information than the bookmark itself and which would 'pop up' on the display when the mouse if hovered over it , as is the case with the KiWi WEB SDR.

This would allow more compact bookmarks to be added with additional information hidden until required. Likewise could the bookmark colour be chosen by the admin in order to differentiate between different users of the spectrum.

In order to make maintenance of the bookmarks easier would it be possible to provide a csv import / export function in order to facilitate external editing and sorting, plus the import of other frequency databases.

jketterl, 2021-05-14:
would you please split different features into separate issues and use the issue template? it doesn't help with the tracking of features if three of them are crammed into one issue.

CSV import has been requested before, but CSV as a format is too generic and required too much additional information to make this feature feasible.

G8JNJ, 2021-05-14:
Hi Jakob,

The fields would be the same as the headings in the bookmark editor (currently Name, Frequency, Modulation) and the values would have to comply with the format in use. Decimal format and number of characters etc.

Any imports would have to follow this simple format, which is easily achieved by bulk editing in Excel or whatever external application is in use. It's up to the user to meet the required criteria rather than trying to make OWR too clever by accommodating different formats.

Being able to export, edit and import back again is a very useful function if you wish to setup lots of bookmark for things like CB or Marine channels, which is very tedious by hand. For example it's very easy to duplicate and then quickly increment channel numbers in Excel, but it takes a long time in by other more manual methods.

jketterl, 2021-05-14:
I'm sorry, but even when assuming that the format is agreed on, there's still way too much that can go wrong with a CSV file and Excel. I'd still need to build a system to validate all the inputs, and then support users with the errors they would receive.

G8JNJ, 2021-05-14:
OK, I assume it will still be possible to manually edit the bookmark file as was the case with the previous JSON file, in which case I could still use some external process to generate channelised markers.

jketterl, 2021-05-14:
The JSON files are not meant to be edited manually, and even though I won't try to prevent you from doing so, I'll have to cut my support short if anything goes wrong in the process.

It seems like you require certain features when editing the bookmarks that are present in Excel. I suppose it would be better to discuss those. I can however not allocate unlimited resources to improve the bookmark editor. Things would look different if there was more developers to work on the project.