jketterl/openwebrx #243: OWR provides a new bookmark editor - request to add notes field

G8JNJ, 2021-05-14:
OWR provides a new bookmark editor.

This permits easy editing and the addition of new frequency markers.

Would it be possible to add a new 'Notes' field, which could contain more information than the bookmark itself and which would 'pop up' on the display when the mouse if hovered over it , as is the case with the KiWi WEB SDR.

This would allow more compact bookmarks to be added with additional information hidden until required. Likewise could the bookmark colour be chosen by the admin in order to differentiate between different users of the spectrum.

This would facilitate the display of more closely packed bookmarks whilst making more information available to the users.

jketterl, 2021-05-14:
As for the colors: I have been thinking about a tagging system. I have already ported the bandplan tags that I'm using on receiverbook for this purpose.

The idea is that instead of assigning colors to individual bookmarks, you would just add them to a specific tag. That tag would then provide the color when rendering the bookmark.

G8JNJ, 2021-05-14:
The idea is that instead of assigning colors to individual bookmarks, you would just add them to a specific tag. That tag would then provide the color when rendering the bookmark.

I think that would be an improvement and provide some degree of commonalty between different OWR websdr's.

Perhaps also provide a small number of user definable bookmarks to deal with the inevitable edge cases.

jketterl, 2021-05-14:
OpenWebRX had a set of default bookmarks as an example config up until version 0.20. I have removed those after discussing this on our Discord server: the feedback I got there was that they were of no use and users were annoyed that they had to remove them.

G8JNJ, 2021-05-14:
This was a problem with the KiWi too.

The solution was to provide a very basic file with some 'universal' bookmarks to get folks started which could be edited to suit, or another blank one that could be used to start from scratch.

Personally I think it's a good idea to include at least a few as a starting point and guide, and the new editor makes it much easier to remove them if required. 

Perhaps a tick box selection and multiple delete or create copies would be a useful addition ?

jketterl, 2021-05-14:
I have no interest in bouncing between individual interests here, and I don't have the time to come up with a list of "universal" bookmarks, if something like that even exists. The sample made sense to have an example of the file format. With the new bookmark editor, this is no longer necessary.