jketterl/openwebrx #244: First and last 2 minute WSPR time interval are not decoded when using scheduler.

erikdewildt, 2021-05-14:
Hi,

First of all thanks for this amazing project! I'm glad to see the new release supports uploading spots to WSPRnet!

**Describe the bug**
The first and last 2 minute intervals are not decoded and/or send to WSPRNet. All intermediate time intervals are decoded as expected.

**To Reproduce**
I've created a schedule which switches bands every 4 minutes. I've noticed no WSPR spots got decoded until a user starts a session and the SDR is on a specific band for a longer period of time. I've then generated a new schedule and increased the time on each band to 6 minutes. This resulted in partial decoding of the received WSPR beacons.

What i observe is that all decoded spots are only from the second interval of each schedule entry. No WSPR beacons from the first and last 2 minute interval of a schedule are decoded.

My schedule looks like this (partially):

... ... ...
1200 - 1206   12 M
1206 - 1212   15 M
1212 - 1218   17 M
1218 - 1224   20 M
... ... ...

The decodes spots are all from 1202, 1208 1214, 1220. As the spots should be completely random i would also expect spots at 1200 or 1206 or 1212 etc. I've decoded about 150 spots the last couple of hours and was not able to see any spots from the first or last 2 minute time interval for each scheduled entry.

**Expected behavior**
As the schedule switches band on even minutes which is the same as the start of a WSPR transmission i would expect WSPR transmissions to be decoded for all 2 minute time intervals within a schedule. My current configuration only decodes 33% of all WSPR transmissions.

**Installation method**
Raspberry Pi SD card image

**Versions**
1.0

**Log messages**
-

**Additional context**
If there are any technical limitations for this then I'm totally fine with it. But it would be super nice to be able to switch bands frequently while decoding WSPR spots as much as possible.

Thanks again!
Regards,

Erik, PE1PUX

jketterl, 2021-05-14:
Sorry, but the automatic scheduling system has its limits. The startup of background services is delayed on purpose so that when the receiver is running with a single user, a simple refresh of the receiver page does not cause the immediate spawn of a bunch of decoder processes, just to take them back down a second later. This causes the first decode to be truncated and probably unusable.

The trailing decodes are probably interrupted before they can finish. That's a simple race condition.

In general, the scheduler was implemented as a means to pick certain bands to adopt to periodic changes of conditions over the course of a day, but not to implement band hopping in short-term intervals. The way the implementation basically means that the more you switch, the more decodes will be lost.

erikdewildt, 2021-05-14:
Hi Jakob,

Thanks for your quick response! 

I understand what you are saying. I'm probably more using the application as a skimmer then a WebSDR at the moment. 

Anyway, the band hopping gives some insight in band openings. For the modes with shorter time intervals like FT8/4 missing some transmissions is not such a big issue as they are in much greater numbers.

Perhaps i can add a second SDR receiver in the future so i can monitor two band at the same time and thus limit band hopping.

Have a nice weekend!
Regards,

Erik, PE1PUX

closed 2021-05-14 by erikdewildt