jketterl/openwebrx #245: Squelch on/off toggle

G8JNJ, 2021-05-15:
**Feature description**
Would it be possible to modify the behaviour of the squelch button (or by some other method) so that it toggles between on and off without affecting the Squelch threshold ?

Currently if I wish to disable the squelch to listen to a weak signal I have to move the squelch slider to zero and back again to it's original position, or press the squelch button to restore it. However if a signal is present when the squelch is restored, the new auto squelch threshold is set higher than the signal so it remains muted unless the squelch slider is moved to a lower threshold, or OWR is tuned off frequency and the auto button used to set a more appropriate threshold, and is then tuned back on frequency.

**Target audience**

All users would benefit from the ability to quickly disable the squelch to listen to weak signals, with the ability to quickly restore squelch operation without having to adjust the slider control.

jketterl, 2021-05-15:
This has been discussed before. The squelch does not have an on/off switch, so there can't be a toggle.

https://groups.io/g/openwebrx/topic/77613911

closed 2021-05-15 by jketterl

G8JNJ, 2021-05-15:
Nevertheless it's still a problem, which is why I raised the issue again in a more formal manner.

I had previously suggested one way around the Squelch button press, which was to use a short press to set the squelch level and a longer button press >1second to toggle the squelch action, however on reflection it would perhaps be better if this was the other way around, with a long button press setting the auto level.

Ideally needs a separate button to perform the toggle function, but I appreciate that space is at a premium.

Some suggestions.

- Add a new button

- Use the squelch slider 'knob' to toggle the squelch auto level and use the current squelch button as the on/off toggle.

- Create more panel space for controls.

By grouping all the digital voice modes into a drop-down like the other digital modes. You will eventually run out of panel space anyway if you continue to add more voice digi modes.  Actually this may work better for voice digimodes, as I tend to toggle through them all anyway if I come across a transmission of unknown mode, and it's easier to scroll down a list than have to press individual buttons.

Or toggle between similar modes such as USB, LSB and CW with each press of a single button.

jketterl, 2021-05-15:
Well the thing is... you didn't describe a problem, but instead proposed the same feature that was already turned down before because it doesn't work. I don't know why you expected the result to be different this time. Doesn't matter if it's posted in a more formal manner this time.

G8JNJ, 2021-05-15:
The problem is that it is difficult to quickly turn the squelch on and off to listen to a weak signal that is close to the squelch threshold. If you try using the squelch control on anything other than strong signals, you will quickly appreciate the problem.

The only method that can be used at the moment to remove the squelch and restore it again, is to move the squelch threshold slider, but this is difficult to adjust and requires some degree of accuracy when being set back to the previous value.

I have suggested three different methods of overcoming the problem of how to find space to  incorporate a squelch on / off feature in the web GUI, but there are likely to be many more ways that this could be achieved.

I'm simply registering the request on Github in the hope that it could be resolved at some future time, or may be considered when the control panel is re-vamped. 

I genuinely appreciate the effort you have put into developing OWR and I do not wish to cause you additional aggravation.