jketterl/openwebrx #251: UHD device not founf

peterg10, 2021-06-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Installation method**
How did you install OpenWebRX? (Raspberry Pi SD card image, Debian / Ubuntu packages, Docker image, manually?)

**Versions**
What version of OpenWebRX are you running? (Check on startup, or see `owrx/version.py`. If a `-dev` version is used, ideally state the commit the issue is appearing on)

**Log messages**
Are there any relevant messages relating to the bug in the output / log of OpenWebRX? (On most installations, the log should be available using the command `sudo journalctl -u openwebrx`)

**Additional context**
Add any other context about the problem here.

jketterl, 2021-06-15:
Please use the issue template to provide information about your issue.

closed 2021-06-15 by jketterl

peterg10, 2021-06-15:
I've tried building manually and using the `openwebrx-uhd` (OpenWebRX version v1.1.0-dev) docker image, in both cases no SDR device is found.

The USRP is found with `uhd_find_devices` and `SoapySDRUtill --find`

I see that the notes for the non-uhd docker specify a --device argument: `--device /dev/bus/usb` is there a similar argument needed for UHD devices?

jketterl, 2021-06-15:
Issue is closed. Your text doesn't describe any bug either, so not reopening. Please follow the instructions and post support requests on the mailing list.