jketterl/openwebrx #253: Digiham inactive in feature report despite being installed

hardlifeofapo, 2021-06-22:
**Describe the bug**
Following a clean installation of openwebrx from the Ubuntu repositories into Hirsute, I cannot decode YSF, DMR, etc. This option was available in a previous installation.

Feature report reads "NO" to the right of any item related to digiham.

Adding outputs from several commands below for additional reference:
uname -a
Linux ThinkPad 5.11.0-16-generic #17-Ubuntu SMP Wed Apr 14 20:12:43 UTC 2021 x86_64 x86

sudo apt install digiham 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
digiham is already the newest version (0.5.0~105.gbp3f7ec7).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

sudo apt install openwebrx 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
openwebrx is already the newest version (1.1.0~890.gbp7a4ed3).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

**To Reproduce**
Clean install in a new machine. Follow the instructions in https://www.openwebrx.de/download/ubuntu.php

**Expected behavior**
With digiham installed, digital voice decode should be available

**Installation method**
Ubuntu Hirsute packages. The same behavious happens with the stable and experimental repos.
apt remove --purge used between installations

**Versions**
OpenWebRX version v1.1.0-dev, from the package in the docs:
https://repo.openwebrx.de/ubuntu/ unstable main

jketterl, 2021-06-22:
> Expected behavior
> With digiham installed, digital voice decode should be available

This is no longer true. I can no longer supply mbelib, which makes decoding the AMBE voice codec impossible. There is a new component `codecserver` which has the ability to use AMBE-3000 devices, e.g. USB-Sticks.

OpenWebRX will check if the AMBE codec is available from the codecserver and report this in the feature report in `codecserver_ambe`.

See this message for more information: https://groups.io/g/openwebrx/message/3487

closed 2021-06-22 by jketterl

hardlifeofapo, 2021-06-22:
Thanks for the info. This makes a lot more sense now.