jketterl/openwebrx #255: How to use Soapy_remote and/or rtl_tcp?

Niemand112233, 2021-07-04:
Hi,

where can I configure a connection to soapy_remote or rtl_tcp? I can't find anything in the docs.

jketterl, 2021-07-04:
Please use the groups.io mailing list for support requests. This area is reserved exclusively for issues within the software.

closed 2021-07-04 by jketterl

Niemand112233, 2021-07-04:
This isn't very userfriendly since I don't have an account at groups.io

Niemand112233, 2021-07-04:
> Please use the groups.io mailing list for support requests. This area is reserved exclusively for issues within the software.

This isn't very userfriendly since I don't have an account at groups.io

jketterl, 2021-07-04:
You don't need to have an account there. All you need is a valid e-mail address.

It's not very friendly to expect that I shift the way I provide support around just because it doesn't suit you. I don't see why some individuals think the world around needs to change to match their comfort.