jketterl/openwebrx #256: hpsdr connector UI Settings

dj1an, 2021-07-10:
**Describe the bug**
Gain Setting in Webinterface for hpsdr needs to be int, not float.

**To Reproduce**
Set a rf gain of 30 in Webinterface, will autocorrect and put 30.0 in settings.json

**Expected behavior**
Sets LNA gain between 0 (-12dB) and 60 (48dB) (default 20)

**Installation method**
Manually from git

**Versions**
Newest git

**Log messages**
```
2021-07-10 13:05:43,504 - owrx.source - INFO - Started sdr source: hpsdrconnector --samplerate 384000 --gain 30.0 --radio 192.168.92.82 --frequency 1900000 -p 43219 -c 47929
invalid value "30.0" for flag -gain: parse error
```
**Additional context**
It would be nice to have the remote IP to be configured via Webinterface Settings.
Else the connector would choose the first available Device.
Adding "remote": "192.168.92.82" in settings.json did the job.

73 de DJ1AN, Jan

jketterl, 2021-07-10:
Pinging @jancona for the gain issue since I cannot reproduce this without a device. If it's really an issue, it should be a matter of adding a converter to `int` to fix this.

For the next time: It's generally a good idea to open separate issues for separate requests. The `remote` input mentioned in the "additional context" has nothing to do with the gain. It has been added in #235 so that should no longer be an issue at all. Please check.

dj1an, 2021-07-10:
You're right, it was fixed in the develop branch. 

Works now as expected ;)

closed 2021-07-10 by dj1an

jketterl, 2021-07-10:
The gain, too?

dj1an, 2021-07-10:
Yes. Gain was also set correctly

jketterl, 2021-07-10:
Ah. OK, well then this is truly resolved. I was surprised because I didn't see it fixed in the PR I linked, but I'm not going to complain either :wink: