jketterl/openwebrx #257: Please add MSK144 decoder to UI and background decoding

had-art, 2021-07-11:
**Feature description**
Please add MSK144 decoder into UI and background decoding

The signal can be decoded using WSJT-X.

**Target audience**
Basically this is a very important 50MHz/2m/70cm DX mode used a lot of ham operators for meteor scatter.

With the background decoding enabled we would be able to setup complete multi-protocol monitoring stations.

Thank you!

jketterl, 2021-07-11:
Please read the project backlog! This has been on there for quite some time now, and you're supposed to check that before opening new tickets! I don't understand why it's so hard!

I don't think this will become a reality very soon though for two reasons:
- Contrary to pretty much any other mode in WSJT-X, this does not come as a standalone decoder that can be integrated. That means it has to be reimplemented from scratch, which is way more laborious.
- The mode is pretty rare, as such testing will be tedious.

I'm tempted to close this right away because it doesn't add anything to the already existing backlog entry, but since people can't be bothered to read, I suppose there's no other way than having repeated information all over the place... And just so you know why this is a problem: The more time I spend on managing the project (due to duplicates), the less time I have to implement features.

dm5tt, 2021-08-12:
Hi! I'm also interested in this topic. Maybe some interesting news regarding this topic.

There seems to be a MSK144 stream decoder available:

https://github.com/alexander-sholohov/msk144decoder

Looks like he wrapped his decoder around the existing WSJT-X decoder routines.

jketterl, 2021-12-21:
There's a really stupid problem with that repository... it is using the MIT license, whereas the WSJT-X code is released under GPLv3. As far as my legal understanding goes, that's a violation of the GPL...

jketterl, 2023-03-17:
This has been implemented in develop.

closed 2023-03-17 by jketterl