jketterl/openwebrx #258: hpsdrconnector auto gain

takgr, 2021-07-12:
hpsdrconnector doesn't seem to support automatic gain control
when we set gain to auto, the following error occurs and the device gets unavailable:

"invalid value "auto" for flag -gain: parse error"

jketterl, 2021-07-12:
Please use the issue template.

This has already been fixed in development (see #235) and there's no AGC option available any more. Since you didn't use the issue template, I don't know what version you're on, and am simply going to close this based on the assumption that you're not using the latest code available.

closed 2021-07-12 by jketterl

takgr, 2021-07-12:
I am using the docker version, downloaded the containers yesterday. Sorry I missed the issue template thing.

jketterl, 2021-07-12:
"docker version" is not distinct enough either since all versions of OpenWebRX (development, stable, and even older version going back to 0.18) are available on docker.

takgr, 2021-07-12:
I was using the "stable" version
after upgrading to "latest" all the hpsdrconnector issues seem to have been resolved :)