jketterl/openwebrx #259: FT8 band reported to PSKReporter turns into "Unknown" as soon OpenWebRX switches into background mode

dm5tt, 2021-07-18:
**Describe the bug**
Band shows up as "Unknown" in PSKReporter when running in background decoding mode.

Everything looks fine when its in the foreground

**To Reproduce**
1. Enable FT8 Background decoding in 2m Band
2. Compare spots in PSKReporter when OpenWebRX is in foreground  or  background mode

**Expected behavior**
Showing "2m" band not not "Unknown"

**Installation method**

Docker

**Versions**
2021-07-18 12:33:22,390 - owrx.__main__ - INFO - OpenWebRX version v1.1.0-dev starting up...

Always "latest" from docker.

Ubuntu 21.04, AMD64

**Log messages**
No related error/warnings

**Additional context**

Here you can see the 2m band reported correctly in foreground mode. As soon it switches into the background the band turns into "Unknown"

[image]

Had a look at all old tickets reported here. This one seems to be new.

jketterl, 2021-07-18:
The band (as seen in the screenshot) is not reported by OpenWebRX, in fact the pskreporter API does not even have a "band" field. It is determined by pskreporter based on the reported frequency.

dm5tt, 2021-07-18:
Hmm.. OK. Thank you!

dm5tt, 2021-07-19:
Band reportings reappeared. Looks like this was a PSKReporter hickup.

closed 2021-07-19 by dm5tt