jketterl/openwebrx #261: Digital modes missing after upgrade to v1.1.0

Danie10, 2021-08-03:
**Describe the bug**
DMR, DStar, YSF modes missing on users web interface.

**To Reproduce**
Performed upgrade as per instructions on a Raspberry Pi 4 by using the `sudo-apt-get install openwebrx` option, and rebooted after finished.

**Expected behavior**
To be showing all digital modes as before.

**Installation method**
Was an existing working version. Just ran the install option to perform the upgrade, and also ran the update for OS. It all came up fine afte rthe reboot except those modes were missing. See [](https://openwebrx.gadgeteerza.co.za).

**Versions**
V1.1.0

**Log messages**
No relevant errors - only `Modulation "psk31" is not available`

**Additional context**
I did look to check that codecserver is running and yes I see an active service running.

jketterl, 2021-08-03:
Did you set up your AMBE device in codecserver?

Danie10, 2021-08-03:
I just did the upgrade, and noted that codecserver settings in OpenWebRx only needed a server address if it was a remote one, so mine was blank. No other configs done as I did not see any further instructions. It seems codecserver was installed as it is running as a service, but not sure of what sort of config I'd need to do where for it?

jketterl, 2021-08-03:
You need to tell codecserver how it can access your AMBE device, at the least it needs to know the corresponding tty device and baud rate. Please check `/etc/codecserver/codecserver.conf`.

Danie10, 2021-08-03:
OK thanks let me go and look at that and see if I can set it. I don't think I've missed an upgrade release, but did not recall having to configure that.

Danie10, 2021-08-03:
OK everything in that config file is commented out so obviously not configured. My log shows my device as:

```
Found 1 device(s):
Aug 03 20:06:08 openwebrx openwebrx[420]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
```
I think OpenWebRx was detecting the and no tty device or baud rate settings that I can see.

jketterl, 2021-08-03:
that's the sdr device, that has nothing to do with codecserver.

do you haven an AMBE device? one of the AMBE USB sticks, DVStick or ThumbDV?

Danie10, 2021-08-03:
No sorry receiver only with an RTL-SDR dongle. It was receiving DMR fine though before upgrade. I thought that codecserver is a software device.

jketterl, 2021-08-03:
No, sorry. there's no software decoding for AMBE included any more. The "mbelib" library has been removed due to legal concerns. Please also see: https://groups.io/g/openwebrx/message/3487

Codecserver is just a dispatcher, mainly because AMBE hardware comes in various shapes and sizes, but it doesn't do any actual decoding.

closed 2021-08-03 by jketterl

Danie10, 2021-08-03:
OK sorry then I've missed something big here. Is DMR etc not going to work at all on RTL-SDR dongles then?  That's a change that these users need to note as I have a lot of non-DMR users monitoring our brand new DMR repeater in Cape Town from my OpenWebRx. If so I'll need to try revert to the older version and stay on that maybe?

jketterl, 2021-08-03:
This has nothing to do with the SDR device in use. DMR uses the proprietary AMBE codec, which is available through purchase of a corresponding piece of hardware. The software solution that was used in earlier versions is in conflict with patents, as such I cannot provide any support for that any more. Downgrading is up to you, I cannot keep supporting old versions.

Danie10, 2021-08-03:
Ah OK thanks for making that clear. Yes, I did not pick up clearly that without an additional hardware device the DMR etc would no longer be working at all. What exactly is that piece of hardware so anyone interested can see if they can source it to continue using these modes?

jketterl, 2021-08-03:
DVSI primarily sell the bare chips, they do have USB devices available but I think they'll sell to corporate customers only.

The hardware that's available to end-users, at least to my knowledge, is the DVStick30 and DVStick33 by DVMega and the ThumbDV by NW Digital Radio. There is other solutions available containing said chips, some manufacturers have produced small batches of Raspberry Pi hats, so it's probably best to have a look around the shops.