jketterl/openwebrx #262: Commercial licensing option?

tpimh, 2021-08-17:
I don't think this part of the readme is still relevant.

> OpenWebRX is also available under a commercial license on request. Please contact me at the address <randras@sdr.hu> for licensing options.

Or is it?

jketterl, 2021-08-17:
Please use the groups.io mailing list for general questions. This space should only contain issues with the software.

To be honest, I have no idea if András would still be offering custom licensing. Since the code is still in part original, I am not in a position to change the license, so I have left this paragraph untouched.

tpimh, 2021-08-18:
@ha7ilm could you please comment on this?

ha7ilm, 2021-08-18:
Hi All! It is still relevant to all the code in github.com/ha7ilm/openwebrx
, and thus also to code from the commits that jketterl/openwebrx shares
with that repo.

Dima Golovin ***@***.***> ezt írta (időpont: 2021. aug. 18.,
Sze, 15:31):

jketterl, 2021-08-22:
Thank you, I guess that settles that, the line stays where it is.

closed 2021-08-22 by jketterl