jketterl/openwebrx #264: No Settings Button on openwebrx GUI

anthropic-radio-source, 2021-09-06:
**Describe the bug**
On my openwebrx gui, there is no Settings button shown. The other buttons (Status, Log, Receiver, Map) show and work OK. This happens wheter I am full screen or reduced screen.

**To Reproduce**
It happens all the time even after a full reboot.

**Expected behavior**
To show the Settings button

**Installation method**
How did you install OpenWebRX?  - Raspberry Pi SD card image, 

**Versions**
What version of OpenWebRX are you running? V1.1.0

**Log message

OpenWebRX client log
Author contact: Jakob Ketterl, DD5JFK | OpenWebRX homepage
Support and information: Groups.io Mailinglist
WebSocket opened to ws://192.168.1.195:8073/ws/
Server acknowledged WebSocket connection.
Audio stream is compressed.
FFT stream is compressed.
Web Audio API succesfully initialized, using ScriptProcessorNode API, sample rate: 48000 Hz

**Additional context**

None.

[image]

jketterl, 2021-09-06:
This doesn't look like version 1.1.0 to me. Versions 1.0 and newer will report their version in the client output log like this:

```
Server acknowledged WebSocket connection, OpenWebRX version: v1.1.0
```

Your log is showing the old message format without the version, indicating version 0.20 or older. Please double-check the version and update as required.

anthropic-radio-source, 2021-09-06:
Jakob Thanks for getting back so quick. I’ll check again. I was sure it was 1.1.0 but obviously not! Keep up the good work mate. Regards ARS From: Jakob KetterlSent: 06 September 2021 14:57To: jketterl/openwebrxCc: anthropic-radio-source; AuthorSubject: Re: [jketterl/openwebrx] No Settings Button on openwebrx GUI (#264) This doesn't look like version 1.1.0 to me. Versions 1.0 and newer will report their version in the client output log like this:Server acknowledged WebSocket connection, OpenWebRX version: v1.1.0Your log is showing the old message format without the version, indicating version 0.20 or older. Please double-check the version and update as required.

anthropic-radio-source, 2021-09-06:
Jakob Doh! I thought I was opening the openwebrx on the RPi 4 image I downloaded from the GitHub site, when in fact I was opening the openwebrx on a RPi 4 running the latest Dragon OS image. The Dragon OS version is obviously not the latest openwebrx. Once again, my apologies for wasting your time. Regards ARS From: Jakob KetterlSent: 06 September 2021 14:57To: jketterl/openwebrxCc: anthropic-radio-source; AuthorSubject: Re: [jketterl/openwebrx] No Settings Button on openwebrx GUI (#264) This doesn't look like version 1.1.0 to me. Versions 1.0 and newer will report their version in the client output log like this:Server acknowledged WebSocket connection, OpenWebRX version: v1.1.0Your log is showing the old message format without the version, indicating version 0.20 or older. Please double-check the version and update as required.

jketterl, 2021-09-06:
No problem. I don't know exactly how the installation on DragonOS works or how to update it. If it's using the repository, it should be possible to update using the package manager.

closed 2021-09-06 by jketterl