jketterl/openwebrx #266: Prometheus metrics break on invalid digital mode names

doccodyblue, 2021-09-15:
If a band entry in bands.json is used for background decoding and the entry has a -space- in the name 
it leads to a space in the prometheus metric, forcing the prometheus parser to break.

**To Reproduce**
{
    "name": "11m CB",
    "lower_bound": 26965000,
    "upper_bound": 27405000,
    "frequencies": {
      "js8": 27245000
    },
    "tags": ["public"]
  },

Leads to this metrics entry:
js8call_decodes_11m CB_JS8_total 4

jketterl, 2021-09-15:
Please use the issue template.
@doismellburning: can you confirm this? suggestions on how to mitigate?

jketterl, 2021-11-21:
If I remember correctly, this was addressed by #273, so can we close this?

doccodyblue, 2021-11-22:
Yes, thank you, all working!

closed 2021-11-22 by jketterl