jketterl/openwebrx #268: Using Nginx as front-end proxy

TheBarret, 2021-09-17:
Does someone have a good nginx conf for proxying the websockets,
I tried to make a config in nginx that would proxy websockets as well, yet i get a websocket error.

```
        location / {
                proxy_pass http://192.168.178.24:8073;
        }
        location /ws/ {
                proxy_pass http://192.168.178.24:8073;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "upgrade";
        }
```

Nginx error:
```
[error] upstream prematurely closed connection while reading response header from upstream, client: <ip>, server: <domain>, request: "GET /ws/ HTTP/1.0", upstream: "http://192.168.178.24:8073/ws/", host: "<domain>"
```

Im using the `rtl_tcp -d 0 -a <ip> -p 1029 -s 1024000 -g 29`  for Openwebrx 

Can someone point me out what I did wrong?

jketterl, 2021-09-17:
This is not an issue, nor a feature request. Please use the mailing list for support requests, as advised.

closed 2021-09-17 by jketterl

TheBarret, 2021-09-17:
I was just asking for advice, no need to be a dick about it.

jketterl, 2021-09-17:
I have set up a selection to guide you to the mailing list in such cases, and individual templates for the issues and feature requests to be created here. You have ignored all of that, so don't tell me I'm the dick here.

TheBarret, 2021-09-17:
it's fine, we decided not to use your software. good day sir!

jketterl, 2021-09-17:
Good luck being impolite towards the next guy.