jketterl/openwebrx #271: migrating / import SDR device settings and profiles

cptechnik, 2021-09-26:
**Describe the bug**
manual migrating dont import devices and profiles

**To Reproduce**
put old files in /home/pi

pi@owrx:~ $ sudo mv . /etc/openwebrx
pi@owrx:~ $ sudo service openwebrx stop
pi@owrx:~ $ sudo openwebrx config migrate
Migrating configuration...
Migrating bookmarks...
Migration complete!
pi@owrx:~ $ sudo service openwebrx start
pi@owrx:~ $ reboot

... starting browser with webif and config... devices are not converted...

**Expected behavior**
the device settings should be imported
the profiles should be imported

**Installation method**
Raspberry Pi SD card image - new installation on RasPi4

**Versions**
OpenWebRX version: v1.1.0

**Log messages**
pi@owrx:~ $ sudo journalctl -u openwebrx -g "owrx.config.migration"
Compiled without pattern matching support

**Additional context**
-

jketterl, 2021-09-26:
Why are you moving your home directory to `/etc/openwebrx` before the migration? The configuration should already be there in the first place.

jketterl, 2021-09-26:
What files are in `/etc/openwebrx` after the first step?

cptechnik, 2021-09-26:
pi@owrx:~ $ ls openwebrx
bands.json  bookmarks.json  config_webrx.py  openwebrx.conf  openwebrx.conf.d  __pycache__

jketterl, 2021-09-26:
and where are these files from? more specifically, what version? does the `config_webrx.py` contain the devices you'd like to see?

cptechnik, 2021-09-27:
from old version 0.9 ?... killed/formated... now running v1 how should i look after versionsnumber hmmm...
```
# configuration version. please only modify if you're able to perform the associated migration steps.
version = 3
```
```
sdrs = {
    "rtlsdr1": {
        "name": "RTL-USB-Stick 1 (V3)",
        "type": "rtl_sdr_soapy",
        "device": "2511003001",
        "rf_gain": "29",
        "ppm": 0,
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        # "lfo_offset": 0,
        # rtl_sdr / rtl_sdr_soapy
        "profiles": {
            "10m": {
                "name": "10m-only",
                "center_freq": 28850000,
                "samp_rate": 1700000,
                "start_freq": 29670000,
                "initial_squelch_level": -50,
                "start_mod": "nfm",
            },
            "12m": {
                "name": "12m",
                "center_freq": 24940000,
                "samp_rate": 1000000,
                "start_freq": 24915000,
                "start_mod": "ft8",
                "direct_sampling": 2,
            },
            "6m": {
                "name": "6m",
                "center_freq": 51000000,
                "samp_rate": 2000000,
                "start_freq": 50313000,
                "start_mod": "ft8",
            },

```

... while "apt upgrade" ssh session was killed by wlan interrupt, so i must install new one, new version... i was able to bakup the owrx config files...

ok, config new device is not a big problem, but config about 10 profiles is more work ... but... this is not  what i wanted to say...

i copied old files into new installed system after configured , executed the line to migrate... and nothing happens ...
a single problem of a bad English speaking German old nerd without big programming knowledge, 
or is it a untested subfunction of the standard migrating progress...? ...which works wunderfull...!

jketterl, 2021-09-27:
Well, if I had to guess, I'd say the problem is that the new system already had some device configuration before the migration step, which prevents the migration command from importing the old config from `config_webrx.py`. Any modification to the default devices and profiles would cause this.

The current version of OpenWebRX still contains the backwards compatibility to read the `config_webrx.py`, so as long as your setup does not have any device config, simply restoring the file should bring back your old devices and profiles. The `openwebrx config migrate` step only transfers the settings to the new storage location, but it will not overwrite any previous settings already stored there.

Please keep in mind that this command was created to migrate existing installations after an update, not for migrating data between installations. If you want to use it for the latter, you have to make sure that your new installation is as clean as possible.

cptechnik, 2021-10-03:
ok...  I'll test without having a config.

If this is verified (by yourself) please add this small warning "works only on empty configuration" to the wiki sites, thanks a lot!

jketterl, 2021-11-21:
I'm not sure what wiki pages you're referring to. I know that [this](https://github.com/jketterl/openwebrx/wiki/Migrating-to-OpenWebRX-1.0) mentions the `openwebrx config migrate` command, but it is titled "Migrating to OpenWebRX 1.0", which should make it clear enough that this only covers a version upgrade, but not migrating between machines.

cptechnik, 2021-11-21:
Now (since your post on sep 28) it's clear, it is the migration from old to new configuration... 
And it's only moving to the new destination...
Before this time I didn't know where the configuration is stored...
It's very important to have configuration files to copy from one to another pi ... Also to backup and restore in case of broken pi (not owrx) data ... (by sudden power interrupt) or... Moving bookmarks to a owrx of a another friendly/buddy owrx user.
Meanwhile I found the new place of the config files... I'll try this next time...

closed 2021-11-21 by cptechnik